micropython/micropython #10908: ESP32-C§ revision 0.3 hangs after installing micropython (1.18 or newer)

raps500, 2023-03-02:
I have a ESP32-C3 revision 0.3. I can install micropython (1.18, 1.19.1) but none work:
`esptool.py v4.6-dev
Serial port COM13
Connecting.........
Chip is ESP32-C3 (revision v0.3)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 84:f7:03:a6:19:bc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00161fff...
Compressed 1446992 bytes to 863628...
Wrote 1446992 bytes (863628 compressed) at 0x00000000 in 27.0 seconds (effective 429.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...`

On Thonny I get:
`Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0xe3c
load:0x403ce000,len:0x6dc
load:0x403d0000,len:0x28c4
entry 0x403ce000

Device is busy or does not respond. Your options:

  - wait until it completes current work;
  - use Ctrl+C to interrupt current work;
  - reset the device and try again;
  - check connection properties;
  - make sure the device has suitable MicroPython / CircuitPython / firmware;
  - make sure the device is not in bootloader mode.

`

It seems to hang. Version 1.19.1 or latest nigthly build has the same issues. 

I have 4 of these boards :(

Any idea if support is going to be added if not there or what the problem might be ?

robert-hh, 2023-03-02:
Can you post a picture of the board? Maybe it's missing the two transistor logic. I have such boards as well, and they do not run with Thonny or mpremote, because the require both RTS and DTR to be off. Picocom (linux) or Tealterm (Windows) can do that, but realterm does not work at all Windows systems.

raps500, 2023-03-03:
These are the boards: 
The UART chip is a CH340. On the underside they are marked ESP-C3-32S-Kit. 
[image]