micropython/micropython #10937: PRO CPU has been reset by WDT.

sebert007, 2023-03-05:
I am using the "Generic ESP32-S3 (SPIRAM Octal)" firmware.

When starting the following loop ("transmit" starts an SPI transfer), I get an error.

```
while True:
     transmit(bytearray([0b00000011, 0b11100000]))
     time.sleep_ms(500)

```

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403757d1
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0xf34
load:0x403c9700,len:0xc44
load:0x403cc700,len:0x2c78
entry 0x403c98f4
W (29) boot.esp32s3: PRO CPU has been reset by WDT.
W (29) boot.esp32s3: APP CPU has been reset by WDT.
```

Is it maybe related to this topic here?
https://esp32.com/viewtopic.php?t=26898

Image:
GENERIC_S3_SPIRAM_OCT-20230207-unstable-v1.19.1-854-g35524a6fd.bin

dpgeorge, 2023-03-06:
Please provide a full working example that shows the problem.

sebert007, 2023-03-06:
```
from machine import Pin, SPI
import time

# ESP32-S3-WROOM
SPI_SCK_PIN = 13
SPI_MOSI_PIN = 11
SPI_MISO_PIN = 12
SPI_CS_PIN = 10

cs = Pin(SPI_CS_PIN, Pin.OUT)

spi = SPI(2)

cs.value(1)

def create(bits):
    spi.init(baudrate=SPI_BAUD_RATE,
             polarity=0,
             phase=0,
             bits=bits,
             firstbit=SPI.MSB,
             sck=Pin(SPI_SCK_PIN, Pin.OUT),
             mosi=Pin(SPI_MOSI_PIN, Pin.OUT),
             miso=Pin(SPI_MISO_PIN, Pin.OUT)
             )

def transmit(buffer):
    print(buffer)
    print("Transmitting...")
    cs.value(0)
    spi.write(buffer)
    cs.value(1)

create(10)

while True:
    transmit(bytearray([0b00000011, 0b11100000]))
    time.sleep_ms(500)
```

When I remove the while-loop and call transmit() manually, it works fine.

Also it only appears on the SPI octal mode devices. The quad mode ones (like the ESP32-S3-DevKitC-1-N8R2 / of course using a different micropython flash version) work fine with this code.

Are you able to reproduce the behavior?

jonnor, 2024-09-01:
Does this still happen with latest MicroPython? There have been several improvements for ESP32S3

closed 2025-02-04 by projectgus