micropython/micropython #10938: super() doesn't work with @classmethod

North101, 2023-03-05:
This doesn't work on micropython (rp2-pico-w-20230302-unstable-v1.19.1-915-g2bcd88d55.uf2) but does work on python3
```
class Foo:
  value = 'test'
  @classmethod
  def foo(cls):
    return cls.value

class Bar(Foo):
  @classmethod
  def foo(cls):
    return super().foo() # also tried super(Bar, cls).foo()
```

I get the following error:
`AttributeError: type object 'type' has no attribute 'value'`

vshymanskyy, 2024-08-21:
@dpgeorge is this expected?

dpgeorge, 2024-08-21:
Both of the above tests should be fixed by 233f5ce661d31c8e606ff727ac0c6de27d32a715

@vshymanskyy can you please test your case with latest MicroPython?

vshymanskyy, 2024-08-21:
Thanks! I can confirm the fix.

dpgeorge, 2024-08-21:
Great, thanks for confirming.

closed 2024-08-21 by dpgeorge