micropython/micropython #10947: RP2040: ADC value different in ADC(channel) vs ADC(Pin(#))

scruss, 2023-03-06:
A user on the the Raspberry Pi forum noticed this issue: [ADC channel number vs. GPIO pin number - Raspberry Pi Forums](https://forums.raspberrypi.com/viewtopic.php?p=2087648#p2087330). In short, if an ADC is initialized in this manner:

    adc = ADC(2)

(that is, by ADC channel) it will return a higher reading than if it is initialized by Pin using:

    adc = ADC(Pin(28))

A user who cannot post issues here noted that initializing by channel [leaves the pull-down enabled](https://forums.raspberrypi.com/viewtopic.php?p=2087648#p2087378), while initializing by pin does not. There is example code in the link that demonstrates the issue.

The Raspberry Pi MicroPython introductory documentation uses the channel method, and thus may be producing spurious results for users.

 Version information shown in the REPL:
```
MicroPython v1.19.1-746-gf2de289ef on 2022-12-13; Raspberry Pi Pico with RP2040
```

th3w, 2023-03-06:
Out of  curiosity I have also tried to find the cause today:

In port/rp2/machine_adc.c -> machine_adc_make_new at  line 88 ,  the sdk function adc_gpio_init(channel) will only be called if value of channel is between 26-29. But the channel value  can also be a number between 0 and 4. (4 is a special case because this is the internal ADC_TEMPSENSOR)
So  a given channel value between  0-3 must first translated to gpio number 26-29 before calling adc_gpio_init().

scruss, 2023-03-06:
Thanks for the PR. I suspect we may need to do something for Channel 4, as it has always read far higher than I'd expect. I'm not sure that it's exposed as a Pin number, though

dpgeorge, 2023-03-07:
Honestly, I think the original behaviour is correct and gives the most flexibility:
- you can specify low channel numbers to get access to the raw channel without any configuration, eg `ADC(2)`
- you can specify high channel numbers to initialise the GPIO, eg `ADC(28)`
- you can pass in a Pin to use ADC on that pin, eg `ADC(Pin(28))`

scruss, 2023-03-07:
But where does it say "use the Pin or high channel number or you'll get unexpected readings"? If this is not made utterly clear in how ADC is accessed, it can't be the correct behaviour.

dpgeorge, 2023-03-07:
The problem is, initialising the ADC for GPIO reading when specifying a low channel number may break existing use of the ADC this way.

I'm not exactly sure how the RP2040 works internally, but on other MCUs it's possible to configure a pin for use by some other peripheral (eg I2C, SPI, or more likely, DAC), and then also create an ADC that "passively" samples that pin.  This can allow you to monitor the loading on the pin.

The implementation of `adc_gpio_init()` is the following:
```c
static inline void adc_gpio_init(uint gpio) {
    invalid_params_if(ADC, gpio < 26 || gpio > 29);
    // Select NULL function to make output driver hi-Z
    gpio_set_function(gpio, GPIO_FUNC_NULL);
    // Also disable digital pulls and digital receiver
    gpio_disable_pulls(gpio);
    gpio_set_input_enabled(gpio, false);
}
```
That will cancel any existing peripheral function on the pin (eg I2C, SPI, PIO) which may not be what the user wants.

Doing some quick tests now on a Pico, it's possible to configure a pin in GPIO output mode and also read back the value via the ADC.

So there needs to be a way for the user to create an ADC instance that does not interfere with the GPIO settings associated with that channel.  I agree it's confusing at the moment, channel 0-3 vs 26-29 vs passing in a pin.  Do you have a suggestion how this could be improved?

th3w, 2023-03-07:
The RP2040 datasheet (Page 559) notes that the digital functions of the selected GPIO pin for analog measurement must be disabled before use. So i would expect this behavour as default setting when creating an ADC object.

Maybe when using a already  configured Pin() object as arguement those setting should be taken over without changes. But the Pin Object is currently just used with 'channel = mp_hal_get_pin_obj(source)' to get the gpio pin number and will then be reconfigured short time later with adc_gpio_init(channel).

In addition, it seems to be possible to change the pin configuration any time after creating the ADC() object. This setting would only be resetted to an  adc input with creating a new ADC object for this GPIO pin.

In any case, this should be pointed out in the RP2 Quickrefs in the machine.ADC section.

lurch, 2023-03-07:
The temperature sensor in the RP2040 isn't calibrated (i.e. it can vary slightly from device to device), and it's very sensitive to slight voltage fluctuations. The RP2040 datasheet says:

> As the Vbe and the Vbe slope can vary over the temperature range, and from device to device, some user calibration
may be required if accurate measurements are required.

and

> The on board temperature sensor is very sensitive to errors in the reference voltage. If the ADC returns a value of
891 this would correspond to a temperature of 20.1°C. However if the reference voltage is 1% lower than 3.3V then
the same reading of 891 would correspond to 24.3°C. You would see a change in temperature of over 4°C for a small
1% change in reference voltage. Therefore if you want to improve the accuracy of the internal temperature sensor it
is worth considering adding an external reference voltage.

th3w, 2023-03-08:
I can understand the current API usage now, but this was not obvious to me before. Maybe i missed something somewhere in the documentation?

For some applications, it can be quite useful to simply measure the external voltage on the pin  without changing the function of the pin. For an example, I have here some displays that  outputs analog touch signals on shared data/spi pins. 

I will try to create another PR in the next few days, targeting the rp2 quickrefs in the docs, to add the information about the different Pin configuration,  depending using machine.ADC() with a channel number or a GPIO pin identifier.

dpgeorge, 2024-07-04:
Commit 2b8de7436bbdb54da99bf6bdb79359876fd3bc1d has made it so specifying an ADC channel via a pin number (26 through 29) will initialise the corresponding GPIO.

But specifying an ADC channel 0 through 3 will leave the GPIO untouched.

Documentation about this is added in f36a5654a87838e68fb313272098777f76a94865

closed 2024-07-04 by dpgeorge