micropython/micropython #10951: Compilation failure with GCC 13

stratakis, 2023-03-07:
On Fedora Rawhide we've updated GCC to the version 13. Since then micropython fails to compile with:

```
gcc -I. -Ibuild -I.. -Wall -Werror -Wextra -Wno-unused-parameter -Wpointer-arith -std=gnu99 -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -Os   -fdata-sections -ffunction-sections -fno-asynchronous-unwind-tables -c -MD -o build/main.o main.c
make: Leaving directory '/builddir/build/BUILD/micropython-1.19.1/mpy-cross'
main.c:333:6: error: conflicting types for 'mp_import_stat' due to enum/integer mismatch; have 'uint(const char *)' {aka 'unsigned int(const char *)'} [-Werror=enum-int-mismatch]
  333 | uint mp_import_stat(const char *path) {
      |      ^~~~~~~~~~~~~~
In file included from main.c:32:
../py/builtin.h:59:18: note: previous declaration of 'mp_import_stat' with type 'mp_import_stat_t(const char *)'
   59 | mp_import_stat_t mp_import_stat(const char *path);
      |                  ^~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

dpgeorge, 2023-03-07:
Thanks for the report.  Should be fixed by 32572439984e5640c6af46fbe7c27400c30112ce

closed 2023-03-07 by dpgeorge

stratakis, 2023-03-07:
Thank you for the swift fix!