micropython/micropython #10961: Passing a keyword argument to a subclass of the built-in class "set" causes a crash.

xiaxinmeng, 2023-03-07:
Passing a keyword argument to a subclass of the built-in class "set" causes a crash. See the following example.

test.py
```
class Test(set):
    def __init__(self, a=0):
        self.add(a)

t = Test(a=0)
```

Error message:
>Segmentation Fault(core dumped)

The expected output: work well (behavior on CPython 3.10.8)

how to replicate it:
1. install micropython (unix port)
2. run it under unix port
>->> 'micropython/ports/unix/build-standard/micropython' test.py

Environment:
compiler versions: MicroPython v1.19.1-724-gfb7d21153 on 2023-03-07; linux [GCC 7.5.0] version
operating system: Ubuntu 18.04.6 LTS

stinos, 2023-03-07:
This has nothing to do with keyword arguments, the issue is that in MicroPython you're using a not-yet constructed subclass. E.g.

```
class Test(set):
    def __init__(self):
        self.add(0)
```

has the same issue. But this works as expected:

```
class Test(set):
    def __init__(self):
        super().__init__()
        self.add(0)
```

dlech, 2023-03-07:
Also see https://github.com/micropython/micropython/pull/9997 for a proposed fix to raise an exception instead of seg faulting.

peterhinch, 2023-03-07:
Also see [MP differences](https://github.com/micropython/micropython/wiki/Differences), in particular:

> Subclassing of builtin types is partially implemented, but there may be various differences and compatibility issues with CPython.