micropython/micropython #10963: Block device `readblock(8, 44)` before `ioctl(1)`

pulkin, 2023-03-07:
According to [docs](https://docs.micropython.org/en/latest/library/os.html), a block device interface may implement `ioctl(1)` "initialize the device". In practice, `os.mount(block_device, ...)` reads 44 bytes with offset 8 without bothering to call `ioctl(1)` first. It is probably the auto-detect feature FAT vs littlefs.

robert-hh, 2023-03-07:
The implementation of ioctl(1, arg) is port-specific. At most ports it just returns 0 for internal flash file system. LFS indeed seems to read without calling ioctl(1). Did something fail for you because of the missing init(1, arg) call?

pulkin, 2023-03-07:
LFS actually calls `ioctl(1)`

https://github.com/micropython/micropython/blob/6bb60745bef0fe442dbbcc126beb79c308714816/extmod/vfs_lfsx.c#L85-L87

Answering your question: yes, a user-defined block device such as the one below will fail.
```python
class RAMBlockDev:
    def __init__(self, block_size, num_blocks):
        self.block_size = block_size

    def readblocks(self, block_num, buf):
        for i in range(len(buf)):
            buf[i] = self.data[block_num * self.block_size + i]

    def writeblocks(self, block_num, buf):
        for i in range(len(buf)):
            self.data[block_num * self.block_size + i] = buf[i]

    def ioctl(self, op, arg):
        if op == 1:
            self.data = bytearray(block_size * num_blocks)
        if op == 4: # get number of blocks
            return len(self.data) // self.block_size
        if op == 5: # get block size
            return self.block_size
```

robert-hh, 2023-03-07:
Sorry, I looked at the mount() code. So it's done when the VFSLFS object t is instantiated, before calling mount. FAT calls it also for various actions, like find_volume, mkfs, fdisk. So when you say the lfs calls it, and you have implemented it, what is missing? For instance the typical sequence for mounting a files system (e.g. on rp2) is:
```
import os, rp2
bdev = rp2.Flash()
try:
    vfs = os.VfsLfs2(bdev, progsize=256)
except:
    os.VfsLfs2.mkfs(bdev, progsize=256)
    vfs = os.VfsLfs2(bdev, progsize=256)
os.mount(vfs, "/")
```
So mount() happens after os.VfsLfs2(), which calls ioctl(1, ).

The lines below are from the ramdisk used by the micropython test suite:
```
class RAMFS:
    SEC_SIZE = 512

    def __init__(self, blocks):
        self.data = bytearray(blocks * self.SEC_SIZE)

    def readblocks(self, n, buf):
        # print("readblocks(%s, %x(%d))" % (n, id(buf), len(buf)))
        for i in range(len(buf)):
            buf[i] = self.data[n * self.SEC_SIZE + i]

    def writeblocks(self, n, buf):
        # print("writeblocks(%s, %x)" % (n, id(buf)))
        for i in range(len(buf)):
            self.data[n * self.SEC_SIZE + i] = buf[i]

    def ioctl(self, op, arg):
        # print("ioctl(%d, %r)" % (op, arg))
        if op == 4:  # MP_BLOCKDEV_IOCTL_BLOCK_COUNT
            return len(self.data) // self.SEC_SIZE
        if op == 5:  # MP_BLOCKDEV_IOCTL_BLOCK_SIZE
            return self.SEC_SIZE

try:
    bdev = RAMFS(50)
    uos.VfsFat.mkfs(bdev)
except MemoryError:
    print("SKIP")
    raise SystemExit
```
They allocate the RAM in __init__().

pulkin, 2023-03-07:
Please read carefully the head of this issue. Calling `os.mount(block_device, ...)` (i.e. argument is a block device; NOT a file system instance) will fail for `block_device`s expecting `ioctl(1)`. I added it to the code listing above: `os.mount(RAMBlockDev(512, 16), "/")` will fail not because the block device is blank but because `data` is not a member of `block_device`.

robert-hh, 2023-03-07:
Agreed, os.mount() does not call ioctl(1, .). That matches my code scan. So in your example the cure would be to allocate the RAM in the __init__() method. I do not know whether mount is expected to call ioctl(1, ..). I do not see anything like hat in the documentation you refer to.

pulkin, 2023-03-07:
That's the issue with `mount`; not block devices. Since I am implementing a block device any issue with `mount` is a bug to me. Looking at the [API](https://docs.micropython.org/en/latest/library/os.html#os.AbstractBlockDev.ioctl) I *reasonably assume* that `ioctl(1)` will be called prior to any reads or writes. Currently, what happens with `os.mount(block_device, ...)` is:

1. `read(8, 44)` is issued *probably* to identify the file system
2. FS wrapper (fat or littlefs) consumes the block device and calls `ioctl(1)`
3. Subsequent `read`s are issued to configure the file system

What is the point of `ioctl(1)` mid-way here? How do I proceed if the block device is a black box to me and I do not know the filesystem either so that I have to rely on `os.mount(block_device, ...)` to mount? I probably have to write some sort of wrapper that (a) calls `ioctl(1)` in `__init__` **and** (b) makes sure that subsequent calls to `ioctl(1)` for block devices that already have been initialized produces no effect. This is clearly a design flaw.

robert-hh, 2023-03-07:
As said, I do not know if mount is expected to call ioctl(1, x). The actual documentation does not require the block devices driver to provide it. The other aspect is, that it is not guaranteed that ioctl(1, x) is only called once. In the example to your driver, every call to ioctl(1, x) will create a new device and discard the previous one. That may also not be an intended behavior, even if a RAM block device usually is not used for long time.