micropython/micropython #10969: SDCard garbled filename and can't open.

kurohach, 2023-03-08:
I am a beginner in programming, I have not used GitHub much, and I am Japanese and write using DeepL.
So, please forgive me if my writing is not correct.

First, let me describe this bug in broad strokes.
If a file name in an SD card uses UTF-8 characters, such as Japanese, the file name will be garbled and unusable when the SD card is mounted.

The board, MicroPython version, and IDE used are as follows
```
board   : Raspberry Pi Pico
version : v1.19.1 (2022-06-18)
IDE     : Thonny 3.3.13
```

The following files and directories were prepared on the SD card for testing.
```
1_testDirectoryName
2_テストディレクトリ名
3_testFileName.txt
4_テストファイル名.txt
```
[image]

The code used is as follows
```
from machine import Pin, SPI
import os
import sdcard

cs = Pin(17, Pin.OUT)
mosi = Pin(19)
miso = Pin(16)
sck = Pin(18)

spi = SPI(0, 10_000_000, sck=Pin(18), mosi=Pin(19), miso=Pin(16))

sd = sdcard.SDCard(spi, Pin(17))
os.mount(sd, '/sd')

os.chdir('/sd')

filenames =os.listdir('/sd')

print(filenames)
```
The SDCard.py used is here.
`
https://github.com/micropython/micropython-lib/blob/master/micropython/drivers/storage/sdcard/sdcard.py
`

Here is the result of this code.
```
>>> %Run -c $EDITOR_CONTENT
['System Volume Information', '3_testFileName.txt', '1_testDirectoryName', '2_eX~1', '4_eX~1.TXT']
>>> 
```
Thus, file and directory names using Japanese characters are garbled.

We also tried to access the garbled file by executing the following code in a shell
```
>>> %Run -c $EDITOR_CONTENT
['System Volume Information', '3_testFileName.txt', '1_testDirectoryName', '2_eX~1', '4_eX~1.TXT']
>>> f = open('/sd/'+filenames[1])
>>> f.close()
>>> f = open('/sd/'+filenames[4])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT
>>> 
```
I was able to open the non-garbled file, but not the garbled file.

[image]

An interesting point is that Thonny shows the garbled files as directories.
I also tried to chdir the garbled directory and this file respectively, but the same error occurs.

There is another interesting point.
That is that it works fine in CircuitPython.

Using CircuitPython v8.0.0, we ran nearly identical code as follows.
```
import os
import board
import sdcardio
import storage
import busio

sd = sdcardio.SDCard(busio.SPI(MOSI=board.GP19, MISO=board.GP16, clock=board.GP18),board.GP17)
vfs = storage.VfsFat(sd)
storage.mount(vfs, '/sd')

os.chdir('/sd')

filenames =os.listdir('/sd')

print(filenames)
```

Here is the result of this code.
```
>>> %Run -c $EDITOR_CONTENT
['System Volume Information', '3_testFileName.txt', '1_testDirectoryName', '2_テストディレクトリ名', '4_テストファイル名.txt']
>>> 
```
[image]

Thus, the file was successfully displayed and successfully used in Thonny.
This difference made me think that there might be a bug in MicroPython's os.mount or sdcard libraries.

Also, and I won't go into details now because it would be too long, but if you connect to Pico and create a file using Japanese on an SD card mounted with MicroPython
This file was causing garbled characters when viewed from Windows. However, I was able to open the file properly, and the strings in the file were normal and not garbled at all.

Therefore, I think this is a bug around the Long File Name of the SDcard.

Unfortunately, I do not yet have the technical skills to solve this. Please help me.

dpgeorge, 2023-03-08:
Thanks for the bug report.

My first thought would be that `MICROPY_FATFS_LFN_CODE_PAGE` needs to be changed to 932 to support Japanese encoding.  Doing that is not easy because it leads to compile errors.

But since CircuitPython seem to support this it would be best to check how it works for them.

@dhalbert do you know how CircuitPython's FAT FS configuration differs to MicroPython such that Japanese characters (UTF-8?) are allowed in filenames?

kurohach, 2023-03-09:
Thank you for your comment.

I'm completely new to this so I may be looking in the wrong place, but I've done a lot of testing and noticed something.

If Unicode characters are used in the filename, it appears to be using the 8.3 filename.
The LFN name should be stored after the 8.3 filename, but I think you are reading that LFN as a directory entry.
Therefore, it seems that it should be a file but is changing to a directory.

I also uploaded an empty directory and an empty txt file to GoogleDrive so that I could do this test.
Please copy this to your SD card.

https://drive.google.com/drive/folders/1n0rggbRrAlMgwr5H_aJAfcNkFsXIR4vl?usp=sharing

kurohach, 2023-03-31:
I was able to deal with it by changing the define in ffconf.h as follows.

```
#define FF_USE_LFN 2

#define FF_LFN_UNICODE 2
```

closed 2023-03-31 by kurohach