micropython/micropython #10970: Support for CH32V307 RISC-V

NeiroNx, 2023-03-08:
Interest to run micropython on China low cost MCU:
[image]
**Features**
* Qingke V4F processor, up to 144MHz system clock frequency.
* Single-cycle multiplication and hardware division. Hardware float point unit (FPU).
* 64KB SRAM, 256KB Flash.
* Supply voltage: 2.5V/3.3V. GPIO unit is supplied independently.
* Low-power modes: sleep/stop/standby.
* Power-on/power-down reset (POR/PDR), programmable voltage detector (PVD).
* 2 general DMA controllers, 18 channels in total.
* 4 amplifiers.
* One true random number generator (TRNG).
* 2 x 12-bit DAC.
* 2-unit 16-channel 12-bit ADC, 16-channel TouchKey.
* 10 timers.
* USB2.0 full-speed OTG interface.
* USB2.0 high-speed host/device interface (built-in 480Mbps PHY).
* 3 USARTs, 5 UARTs.
* 2 CAN interfaces (2.0B active).
* SDIO interface, FSMC interface, DVP.
* 2 IIC interfaces, 3 SPI interfaces, 2 IIS interfaces.
* Gigabit Ethernet controller ETH (built-in 10M PHY).
* 80 I/O ports, can be mapped to 16 external interrupts;
* CRC calculation unit, 96-bit unique ID.
* Serial 2-wire debug interface.
* Packages: LQFP64M, LQFP100.

http://www.wch-ic.com/products/CH32V307.html

Toolchain(Eclipse):
http://www.mounriver.com/download

GitHub Examples:
https://github.com/SoCXin/CH32V307

Boards:
https://aliexpress.ru/item/1005004449629983.html
https://aliexpress.ru/item/1005004350410929.html

Possible to buy at JLCPCB(when pcb+assemble) 3.5$ peice:
https://jlcpcb.com/partdetail/wch_jiangsu_Qin_heng-CH32V307RCT6/C2988083

mattytrentini, 2023-03-08:
Already in progress, see #8637. 😄

niutianhu, 2023-11-05:
老板有没有开发板卖