micropython/micropython #10980: ESP32C3 repl halt when calling machine.reset()

Walkline80, 2023-03-08:
I compiled the firmware with board `GENERIC_C3_USB` and tried to reset it using `machine.reset()` on Thonny's repl command line, which caused the repl to freeze and can only be used normally by press the reset button on board.

I have no idea, is there anyone has encountered the same problem?

robert-hh, 2023-03-08:
Blame it onto Thonny. No problem here without Thonny. Calling machine.reset() properly resets the boards and returns to a REPL prompt.

Walkline80, 2023-03-09:
hi, @robert-hh , I've actually tested with several serial port tools, and the results are all the same, could you please tell me which version of esp_idf you're using now? I want to give it a shot.

btw, I'm using esp32c3 virtual serial port to open the repl, dozens of tests will only succeed several times.

robert-hh, 2023-03-09:
The firmware was built with esp-idf v4.4.2. Version 4.2.x do not compile for the ESP32C3, at least not here. My set-up is a Linux Debian box and Picocom, mpremote or tio as terminal emulator. The test board is an Seeed XIAO ESP32C3 with just a few files in the file system.

Walkline80, 2023-03-09:
Well, @robert-hh you're right, I used esp-idf v4.4 before, now switched to v4.4.3 and rebuild the firmware, it's worked fine...

Thank you, thanks a lot!~

robert-hh, 2023-03-09:
Ok. Good to know and happy, that it works.

closed 2023-03-09 by robert-hh