micropython/micropython #10984: Socket.accept() hangs indefinitely in the latest build (v1.19.1-939)

vlad2305m, 2023-03-08:
minimal code (from the tutorial):
```python
import network
import socket
from time import sleep
from picozero import pico_temp_sensor, pico_led

ssid = '[fill in]'
password = '[fill in]'

def connect():
    # Connect to WLAN
    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)
    wlan.ifconfig(('192.168.1.16', '255.255.255.0', '192.168.1.1', '8.8.8.8'))
    wlan.connect(ssid, password)
    while not wlan.isconnected():
        print('Waiting for connection...')
        sleep(1)
    ip = wlan.ifconfig()[0]
    print(f'Connected on {ip}')
    return ip

def open_socket(ip):
    # Open a socket
    address = (ip, 80)
    connection = socket.socket()
    connection.bind(address)
    connection.listen(1)
    return connection

def webpage(temperature, state):
    # Template HTML
    html = f"""
<!DOCTYPE html>
<html lang="en">
<head>
    <meta charset="UTF-8">
    <title>Title</title>
</head>
<body>
<form action="./lighton">
    <input type="submit" value="Light on" />
</form>
<form action="./lightoff">
    <input type="submit" value="Light off" />
</form>
<p>LED is {state}</p>
<p>Temperature is {temperature}</p>
</body>
</html>
            """
    return str(html)

def serve(connection: socket.Socket):
    # Start a web server
    print("client!?")
    state = 'OFF'
    pico_led.off()
    while True:
        print(connection)
        (client, addr) = connection.accept()
        print("client!", addr)
        request = client.recv(1024)
        request = str(request)
        try:
            request = request.split()[1]
        except IndexError:
            pass
        if request == '/lighton?':
            pico_led.on()
            state = 'ON'
        elif request == '/lightoff?':
            pico_led.off()
            state = 'OFF'
        temperature = pico_temp_sensor.temp
        html = webpage(temperature, state)
        client.send(html)
        client.close()

ip = connect()
serve(open_socket(ip))
```
on build 933, the above code works, but on build 939 it gets stuck at connection.accept(), not sending the webpage to my browser.

dpgeorge, 2023-03-08:
I assume you are running this code on a Pico W board?

dpgeorge, 2023-03-08:
I tested the above code with a Pico W and firmware `MicroPython v1.19.1-939-ga234aa44f on 2023-03-09; Raspberry Pi Pico W with RP2040` and it works OK for me.

dpgeorge, 2023-03-08:
BTW, which tutorial did you get this code from, can you please link to it?

vlad2305m, 2023-03-09:
Yes, Pico W. https://projects.raspberrypi.org/en/projects/get-started-pico-w/3
Works fine with today's build (941). Thank you for your time!

closed 2023-03-09 by vlad2305m