micropython/micropython #10998: SPI: wrong mosi pull-up / pull-down state / can't change settings

sebert007, 2023-03-10:
It seems that an SPI instance (SPI, not SoftSPI) switches the pull-up resistor on for MOSI by default.

I am using the following code:

```

miso_pin = Pin(SPI_STD_MISO_PIN, Pin.IN, Pin.PULL_DOWN)

spi_std = SPI(SPI_STD_DEVICE,
              sck=Pin(SPI_STD_SCK_PIN),
              mosi=Pin(SPI_STD_MOSI_PIN),
              miso=miso_pin
              baudrate=SPI_STD_BAUD_RATE,
              polarity=SPI_STD_POL,
              phase=SPI_STD_PH,
              bits=SPI_STD_BITS,
              firstbit=SPI_STD_FIRSTBIT)
```

Just for information the variables:
```
SPI_STD_DEVICE = 1
SPI_STD_SCK_PIN = 8
SPI_STD_MOSI_PIN = 6
SPI_STD_MISO_PIN = 7
SPI_STD_BAUD_RATE = 1_000_000
SPI_STD_POL = 0
SPI_STD_PH = 0
SPI_STD_BITS = 8
SPI_STD_FIRSTBIT = SPI.MSB
```

The MOSI pin pulls high on my oscilloscope after the SPI instantiation. This is really bad, since SPI devices normally are not able to sink current. This means, that MOSI will remain pulled up and data transmission to the master does not work.

It seems that the Pin.PULL_DOWN parameter is overwritten or not taken into account.

As a result it is not possible to use SPI without the pull-up resistor switched on.

SoftSPI doesn't seem to show this behavior but is not an alternative to the hardware SPI.

robert-hh, 2023-03-10:
Which boards and firmware version do you use?

sebert007, 2023-03-10:
Sorry, forgot to mention.

MicroPython v1.19.1-932-g6bb60745b on 2023-03-07; ESP32S3 module (spiram) with E
SP32S3

I am using ESP32-S3 with QUAD SPIRAM (not octal).

For default, it would probably be best to neither have a pull-up nor a pull-down enabled, just neutral (I don't know if this is possible -- but if not, pull-down would be the right choice).

robert-hh, 2023-03-10:
SPI MOSI and SCK outputs are driven at low impedance either high or low in order to achieve a high transfer rate. There is no neutral state. Looking through the SPI API of the espressif SDK, I see no option to set the MOSI level at inactive times. Only SCK quiet level and SCK phase can be selected.
You may however be able to configure the MOSI pin as GPIO and reconfigure it to SPI only when needed. But then there would still be s short time outside tansmission when MOSI is high.

robert-hh, 2023-03-10:
I'll move this topic to discussions, since it does not seem to be a bug.