micropython/micropython #10999: SoftSPI: Implement 'bits' and 'firstbit' parameters

sebert007, 2023-03-10:
SPI offers both this parameters to specify. SoftSPI rejects both of them, although they are really useful (and needed).

But first probably this issue should be fixed along with it.

#5225

robert-hh, 2023-03-10:
SoftSPI allows to use these parameters at the constructor, but only allow 8 for bits and SPI.MSB for firstbit. So practicall it does not support them.  Both look easy to implement (see micropython/drivers/bus/softspi.c and micropython/extmod/machine_spi.c), but someone has to do it and test it.

robert-hh, 2023-03-11:
> But first probably this issue should be fixed along with it. https://github.com/micropython/micropython/issues/5225

Maybe there were to many opinions in that discussion, but no conclusion. And then it sticks.