micropython/micropython #11010: Docker image is missing `mip`

BenjaminEHowe, 2023-03-11:
[The official docker image](https://hub.docker.com/r/micropython/unix) (1.19.1) appears to be missing the `mip`  module:
```
$ micropython-dev -m mip install urequests
ImportError: no module named 'mip'
```

See [an example run via GitHub Actions](https://github.com/BenjaminEHowe/micropython-docker-mip/actions/runs/4394173624/jobs/7695064229#step:3:5).

peterhinch, 2023-03-12:
AIUI `mip` was released after 1.19.1 and hence is available only in nightly builds. Roll on 1.20!

mattytrentini, 2023-03-12:
Yes, I've only (manually) published docker images for releases; the intent is to publish with every commit to master. It's on my list for post-1.20...

jonnor, 2024-08-24:
I see both 1.22 and 1.23 on https://hub.docker.com/r/micropython/unix/tags - so this should be fixed now?

closed 2024-08-24 by robert-hh

BenjaminEHowe, 2024-08-24:
Confirming that this works for v1.23.0...

```
sudo docker run -ti --rm micropython/unix:v1.23.0 sh
# micropython -m mip install urequests
Installing urequests (latest) from https://micropython.org/pi/v2 to /root/.micropython/lib
```

Thanks folks!