micropython/micropython #11022: Hey ! I think I have the same issue and none of the answers has worked.

AlexandreLarribau, 2023-03-13:
Hey ! I think I have the same issue and none of the answers has worked. 

I am listenning 5 pins. And I try to guess which has raised the interrupt. 

p1 = machine.Pin(1, machine.Pin.IN)
p2 = machine.Pin(2, machine.Pin.IN)
p4 = machine.Pin(4, machine.Pin.IN)
p5 = machine.Pin(5, machine.Pin.IN)
p6 = machine.Pin(6, machine.Pin.IN)

def handle_pwm_interrupt(pin):
  
    print("the pin is : ")
    print(pin)

p1.irq(trigger=machine.Pin.IRQ_FALLING | machine.Pin.IRQ_RISING, handler=handle_pwm_interrupt)
p2.irq(trigger=machine.Pin.IRQ_FALLING | machine.Pin.IRQ_RISING, handler=handle_pwm_interrupt)
p4.irq(trigger=machine.Pin.IRQ_FALLING | machine.Pin.IRQ_RISING, handler=handle_pwm_interrupt)
p5.irq(trigger=machine.Pin.IRQ_FALLING | machine.Pin.IRQ_RISING, handler=handle_pwm_interrupt)
p6.irq(trigger=machine.Pin.IRQ_FALLING | machine.Pin.IRQ_RISING, handler=handle_pwm_interrupt)

when I do this the display is "Pin(GPIO4, mode=IN)". 
I tried pin.name(), pin.id(), pin.id and each time i get :"File "<stdin>", line 16, in handle_pwm_interrupt
AttributeError: 'Pin' object has no attribute 'name'" 

And with id(pin) i get "268703464" which doesn't help me much to know which pin it is. 

Do you have advices ?

_Originally posted by @AlexandreLarribau in https://github.com/micropython/micropython/issues/2612#issuecomment-1466308346_
            
[image]

rkompass, 2023-03-13:
You have the pin object in the interrupt handler. So you should be able to compare like:
```
if pin == p1:
    do_something1()
elif pin == p2:
    do_something2()
elif pin == p3:
    do_something3()
... 
```
and so on. Getting the pin number from the pin object is a known cumbersomeness in Micropython: You have to extract it from the string representation. Like:
`gpio_id = int(str(pin).split(", ")[0][4:])` which worked on the RPi Pico.
For the RP2 I found another way, that is not simpler though, iirc.

I see now I repeated old answers/advices.

In the headings you state: **None of the answers has worked.**

Please elaborate what you tried and how it didn't work.

Perhaps you wanted to urge on this problem (is this correct English?). If you wanted to start a query on that feature (having a pin.id() function) I would support it. But consider: Pins are sometimes numbers, sometimes letters, ...

robert-hh, 2023-03-13:
I'll convert that to a discussion, which is the better place to deal with it.