micropython/micropython #11025: Firmware 1.19.1 on NUCLEO-WB55 is failing on BLE functions

Yannick-Marietti, 2023-03-13:
Hello

MicroPython Firmware :  1.19.1
Board : STMicrolectronics NUCLEO-WB55

Since recent STMicroelectronics releases of _stm32wb5x_BLE_HCILayer_fw.bin_, available here :

https://github.com/STMicroelectronics/STM32CubeWB/tree/master/Projects/STM32WB_Copro_Wireless_Binaries/STM32WB5x

all BLE scripts run on NUCLEO-WB55 boards are failing with message :

`tl_ble_wait_resp: timeout.`

Please help !
 I did not found any way to restore BLE functions, even while downgrading _stm32wb5x_BLE_HCILayer_fw.bin_ to rev 1.13 which was the one used used for MictoPython firmware developement.

Best regards

Yannick

dpgeorge, 2023-03-16:
I tried recently with HCILayer firmware from v1.14.0 of the STM32CubeWB repository and it works with the latest MicroPython commit (ie after the stable 1.19.1 release).

Sometimes after programming the FUS/WS and/or MicroPython firwmare you can get `tl_ble_wait_resp: timeout` errors.  A full power cycle will fix that case.

Yannick-Marietti, 2023-03-16:
Hello Damien

Thanks for your reply !

What do you mean with _"A full power cycle will fix that case."_ You suggest I have to switch on/off the board after programming the FUS/WS and/or MicroPython firwmare ?
**I am very interested on that point as I spent many hours searching a workaround for this nasty bug**.

I had several discussions since with my colleagues in ST working on NUCLEO-WB55.
There is another incoming problem : latest releases of the NUCLEO-WB55 (the ones sold in cardboard boxes) are factory programmed with **a BLE stack stripped from HCI interface**.

This is why I spent a lot of time writing this tutorial (in french) explaining how to reprogram the right BLE stack for MicroPython:

https://stm32python.gitlab.io/fr/docs/tools/cubeprog/cube_prog_firmware_ble_hci

Best regards

Yannick

jonnor, 2024-09-02:
Is this problem still relevant for latest MicroPython? It seems that at least with MicroPython 1.19, the issue was not reproducible.

Yannick-Marietti, 2024-09-03:
Hello Jordy

Thanks !
I’m going to check, but it seems yes.
The point is ST changed at factory the RF part of the STM32WB55 SoC firmware, removing HCI interface on latest NUCLEO-WB55 boards. So you’ve got to reload an HCI RF firmware it using STM32cube programmer this way (sorry its written in French) :

https://stm32python.gitlab.io/fr/docs/tools/cubeprog/cube_prog_firmware_ble_hci

Best regards

Yannick

ST Restricted