micropython/micropython #11026: RP2: Poll on usys.stdout.buffer blocks

smurfix, 2023-03-14:
```
import uasyncio
def can_write(s):
    yield c._io_queue.queue_write(s)

async def write_stdout(data):
    await can_write(usys.stdout.buffer)
    usys.stdout.buffer.write(data.encode())

uasyncio.run(write_stdout('hello\n'))
```
This blocks indefinitely on RP2.

The corresponding code for reading works.

dpgeorge, 2023-03-16:
Thanks for the report.  Inspecting the code it seems that `mp_hal_stdio_poll()` only implements `MP_STREAM_POLL_RD` handling for stdin handling (USB CDC and REPL UART incoming streams).  This is also the case for other ports using TinyUSB, eg mimxrt and samd.

To fix, it's easy enough to support `MP_STREAM_POLL_WR`.  But what also needs to be done is for `sys.stdout.buffer.write(data)` to only write data that it can, not block, and return the number of bytes actually written.  That's a bit tricky in the face of `os.dupterm()`.

What's the reason to use `sys.buffer.stdout` in this case?  Is it because you want to write to the USB serial output as part of an asyncio app?

smurfix, 2023-03-16:
> That's a bit tricky in the face of os.dupterm().

Right, but should be possible when there's just one stream.

> What's the reason to use sys.buffer.stdout in this case? 

I'm running a binary protocol (msgpack, basically) on USB-CDC. There's a multiplexer on the server side and modular code on the RP2, so I can basically run a bunch of code (including LittleFs file system access) in parallel.

dpgeorge, 2023-03-21:
See #11091 for a fix.

dpgeorge, 2023-03-23:
Fixed by 38e7b842c6bc8122753cbf0845eb141f28fbcb72

closed 2023-03-23 by dpgeorge