micropython/micropython #11030: mpy-cross version error

SaraEfabless, 2023-03-14:
Hi folks, anyone can help me with this error?
saran@SaraNs-MacBook-Pro ~ % git clone https://github.com/efabless/caravel_board.git
fatal: destination path 'caravel_board' already exists and is not an empty directory.
saran@SaraNs-MacBook-Pro ~ % cd caravel_board/firmware_vex/nucleo
saran@SaraNs-MacBook-Pro nucleo % make setup
python3 -m pip install rshell
Requirement already satisfied: rshell in /opt/homebrew/lib/python3.11/site-packages (0.0.31)
Requirement already satisfied: pyserial in /opt/homebrew/lib/python3.11/site-packages (from rshell) (3.5)
Requirement already satisfied: pyudev>=0.16 in /opt/homebrew/lib/python3.11/site-packages (from rshell) (0.24.0)
python3 -m pip install mpy-cross
E**RROR: Could not find a version that satisfies the requirement mpy-cross (from versions: none)
ERROR: No matching distribution found for mpy-cross**
make: *** [setup] Error 1
saran@SaraNs-MacBook-Pro nucleo % python3 -m pip install mpy-cross
_ERROR: Could not find a version that satisfies the requirement mpy-cross (from versions: none)
ERROR: No matching distribution found for mpy-cross_
saran@SaraNs-MacBook-Pro nucleo % pip3 --version
pip 23.0.1 from /opt/homebrew/lib/python3.11/site-packages/pip (python 3.11)
saran@SaraNs-MacBook-Pro nucleo % mpy-cross --version
MicroPython v1.19.1 on 2023-03-13; mpy-cross emitting mpy v6

SaraEfabless, 2023-03-14:
> I believe you are running into https://gitlab.com/alelec/mpy_cross/-/issues/15 if you have an Arm-based Mac.
>
> It's probably not a drop-in replacement, but https://pypi.org/project/mpy-cross-v6 has a binary wheel for Arm Macs.

Hi Dave, 

Thanks for your fast response. As the terminal lines shows I do have the latest may-cross-v6 installed, but I’m still getting the error of “the requirement version is not satisfied” which is confusing. 

Regards. 
Sara.

dlech, 2023-03-14:
`rshell` depends on the `mpy-cross` package, so the `rshell` package would need to be modified to changed the dependency.

jonnor, 2024-08-24:
This appears to be a rshell problem, not in core MicroPython. Proposing to close this issue.

closed 2024-08-24 by robert-hh