micropython/micropython #11032: The posix implementation of os.listdir() filters all files starting with two periods

jeremysrand, 2023-03-15:
I am using the UNIX port of micro python, v1.19.  If I do the following in the shell:

`$ touch .testing ..testing ...testing`

And then run this from the REPL or a script in the same directory:

```
import os
for file in os.listdir():
    print(file)
```

The output will show .testing but not the files ..testing or ...testing.

The problem is caused by this line in extmod/vfs_posix.c:

```
        if (fn[0] == '.' && (fn[1] == 0 || fn[1] == '.')) {
            // skip . and ..
            continue;
        }
```

This code doesn't check that fn[2] is a NUL char.  So, as long as the first two chars are a period, the file will be skipped.

dlech, 2023-03-15:
Sounds like an easy fix. Would you like to make a pull request?

jeremysrand, 2023-03-15:
Sure.  It does look like a trivial fix.  I will work on a pull request.

dpgeorge, 2023-03-21:
Fixed by d677023b3decec2f10d6b00b84b34032db4a0fd4

closed 2023-03-21 by dpgeorge