micropython/micropython #11033: The posix implementation of os.ilistdir() can leak the DIR pointer

jeremysrand, 2023-03-15:
I am using the UNIX port of micro python, v1.19.  If I execute the following in a directory with two or more files, then the closedir() function does not get called and a DIR pointer is leaked:

```
import os
for entry in os.ilistdir():
    print(entry)
    break
```

The call to closedir() is in vfs_posix_ilistdir_it_iternext() in extmod/vfs_posix.c.  However, this is only called when the iterator reaches the end of the list:

```
        struct dirent *dirent = readdir(self->dir);
        if (dirent == NULL) {
            closedir(self->dir);
            MP_THREAD_GIL_ENTER();
            self->dir = NULL;
            return MP_OBJ_STOP_ITERATION;
        }
```

But, if instead the script terminates iteration early, closedir() is not called.  This leads to a memory leak of the self->dir pointer and also a file descriptor for the directory being iterated upon.  I have confirmed using lsof that every time I do execute and break out of the for loop above, another file descriptor for the directory is opened and remains open.

Note that os.listdir() does not suffer from this leak because it internally always iterates over the entire directory using os.ilistdir() and returns its value as a list.  So, breaking out of a for loop on os.listdir() does not cause a leak.

dlech, 2023-03-15:
Should already be fixed by https://github.com/micropython/micropython/commit/4e0964b59f44b25ebaa5239f9ea4273b804ebe64.

jeremysrand, 2023-03-15:
Agreed.  Sorry.  I should have looked at the latest code more closely and seen the finalizer.  Thanks!

closed 2023-03-15 by jeremysrand