micropython/micropython #11042: ESP32S3R8 - 16MB Flash - 8MB PSRAM?

PsuFan, 2023-03-16:
Is MicroPython supported on this? https://www.amazon.com/gp/product/B09J112YR7/

I can't get it to run... updated to esptool 4.5.1. Mostly boot loops or no serial hardware detected...

PsuFan, 2023-03-18:
I was able to get a circuitpython with 8M Flash - 8M PSRAM running. I really don't like whats happened to circuitpython, libraries have been removed, they took out machine and webrepl. I haven't been able to find any documentation on what happened to webrepl. Still trying to get MicroPython working...

przemoc, 2023-03-19:
Regarding support for [LILYGO® T-Display-S3](https://www.lilygo.cc/en-pl/products/t-display-s3), i.e. development board equipped with:

- [ESP32-S3](https://www.espressif.com/en/support/documents/technical-documents?keys=&field_type_tid%5B%5D=842) with [Xtensa® dual-core 32-bit LX7 microprocessor](https://www.cadence.com/en_US/home/tools/ip/tensilica-ip/tensilica-xtensa-controllers-and-extensible-processors/xtensa-lx-processor-platform.html) ([ISA summary](https://www.cadence.com/content/dam/cadence-www/global/en_US/documents/tools/ip/tensilica-ip/isa-summary.pdf)), specifically ESP32-S3R8 with 8 MB In-Package PSRAM
- [GigaDevice GD25Q128E](https://www.gigadevice.com/product/flash/product-series/spi-nor-flash/gd25q128e) 16 MB SPI Flash (GigaDevice GD25Q128ESIG on the photo)
- color 1.9" ST7789 170x320 LCD screen (non-touch version)
- 2 programmable buttons

[image]

You may want to check their repo:

- https://github.com/Xinyuan-LilyGO/T-Display-S3

which for MicroPython links to:

- https://github.com/Xinyuan-LilyGO/lilygo-micropython

I consider obtaining this board for myself (but having some issues with AliExpress closing order allegedly due to account security concerns, which doesn't make any sense...).

PsuFan, 2023-03-19:
Wow I had no idea they were compiling their own Micropython, thanks! I’ll have to check it out. I have some of theirs with the 1.14 inch screen that work well. Do they have Amazon over there?

przemoc, 2023-03-20:
You're welcome.

I have local version of Amazon, so I can find sort of equivalents for relevant deals from .com like:

- [LILYGO T-Display-S3 ESP32-S3 1.9 inch ST7789 LCD Display TTGO Development Board](https://www.amazon.com/LILYGO%C2%AE-T-Display-RP2040-Raspberry-Development/dp/B09J112YR7) or
- [LILYGO T-Display-S3 ESP32-S3 LCD Display TTGO Development Board Normal Screen with Shell](https://www.amazon.com/LILYGO-T-Display-S3-ESP32-S3-Development-Soldered/dp/B0BRTT727Z),

but they're more expensive than AliExpress one, even when not counting welcome deal:

- [LILYGO® T-Display-S3 ESP32-S3 1.9 inch ST7789 LCD Display Development Board WIFI Bluetooth5.0 Wireless Module 170*320 Resolution](https://www.aliexpress.com/item/1005004496543314.html))

Mostly due to delivery cost (even though this delivery cost in local Amazon is still noticeably cheaper than from global/US Amazon). AliExpress has free shiping (albeit slow). And if I'm in no rush to obtain the board, I don't like to waste money.

LILYGO beside having their own [store](https://www.lilygo.cc/), have their presence in both [Amazon](https://www.amazon.com/stores/page/0F20442A-0A95-46E6-8FCE-80C6D59434E0) and [AliExpress](https://lilygo.aliexpress.com/store/2090076), so it feels somewhat safer to buy from them than from some random stores on those shopping platforms. But for whatever reason, making order on AliExpress with LILYGO stuff fails everytime as I mentioned last time. Hypothesis is that welcome deal may be geo-limited, but I don't see way to disable this kind of automatic coupon that doesn't show as coupon...

Anyway, this is totally off-topic for MicroPython ticket, and I apologize for that.

jkhax0r, 2023-03-31:
@PsuFan I am running custom hardware with the S3 that has 16MB Flash and 8MB RAM.  It's octal and I saw it won't work if you flash the QSPI quad version.

You must use the GENERIC_S3_SPIRAM_OCT version.  https://micropython.org/download/GENERIC_S3_SPIRAM_OCT/

Let me know how its going for you.  I have been doing battle with the gc.collect() time on it... the large RAM results in very long collect times that caused trouble for my use case.  Debating going to lower RAM to speed things up... as screwed up as that sounds!  

This PR was a big improvement... https://github.com/micropython/micropython/pull/10235

AppsByDavideV, 2023-04-03:
> 

Hi jkhax0r,
I have a chinese board S3N16R8 (octal PSram)  that works fine with the micropython  link you provided . But it shows only 8M flash size  instead of 16. I found a micropython version from Lolin manifacturer for their own board with same configuration but showing full 16M flash available.
Is there a way to have the original Micropython supporting full 16M flash  for the S3N16R8 octal PSram?
Thanks ;)

ma261065, 2023-04-06:
See this - https://github.com/Mythologyli/MicroPython-ESP32-S3-N16R8

AppsByDavideV, 2023-04-09:
> See this - https://github.com/Mythologyli/MicroPython-ESP32-S3-N16R8

Hi, done thanks. I used WSL (Linux under windows) to compile.

ma261065, 2023-04-11:
@AppsByDavideV - note that if you compile for a ESP32-S3, then Micropython interrupts don't work - see https://github.com/micropython/micropython/issues/8488. You have to run menuconfig and set the single core flag (under ComponentConfig->FreeRTOS->RunFreeRTOS only on first Core)

AppsByDavideV, 2023-04-11:
> @AppsByDavideV - note that if you compile for a ESP32-S3, then Micropython interrupts don't work - see #8488. You have to run menuconfig and set the single core flag (under ComponentConfig->FreeRTOS->RunFreeRTOS only on first Core)

Good to know, I will check this too. Many thanks!

nspsck, 2023-06-18:
Hi, @AppsByDavideV @PsuFan maybe this is still relevant for you. 
There is a file called **sdkconfig.board** under this dictionary
`/micropython/ports/esp32/boards/GENERIC_S3_SPIRAM_OCT`
simply change the configuartion to:
```
CONFIG_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_ESPTOOLPY_FLASHSIZE_DETECT=y
CONFIG_ESPTOOLPY_AFTER_NORESET=y

CONFIG_SPIRAM_MEMTEST=

CONFIG_ESPTOOLPY_FLASHSIZE_4MB=
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions-16MiB.csv"
```
then you will get the full 16MiB you paid for. Of cause the board name would be **GENERIC_S3_SPIRAM_OCT**.

AppsByDavideV, 2023-06-18:
Hi nspsck,
it's great!.
Thanks for your effort!

jasmohan-narula, 2024-04-23:
I recently tried building micropython for ESP32-S3-WROOM-1 N16R8

I used https://github.com/Mythologyli/MicroPython-ESP32-S3-N16R8 inside WSL Ubuntu 22.04LTS

While building, I'm getting this error:-
`
In file included from /mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/py/objlist.h:29,
                 from /mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/extmod/modnetwork.c:31:
/mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/ports/esp32/modnetwork_globals.h:38:54: error: 'WIFI_AUTH_WPA3_EXT_PSK' undeclared here (not in a function); did you mean 'WIFI_AUTH_WPA3_ENT_192'?
   38 | { MP_ROM_QSTR(MP_QSTR_AUTH_WPA3_EXT_PSK), MP_ROM_INT(WIFI_AUTH_WPA3_EXT_PSK) },
      |                                                      ^~~~~~~~~~~~~~~~~~~~~~
/mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/py/obj.h:90:67: note: in definition of macro 'MP_OBJ_NEW_SMALL_INT'
   90 | #define MP_OBJ_NEW_SMALL_INT(small_int) ((mp_obj_t)((((mp_uint_t)(small_int)) << 1) | 1))
      |                                                                   ^~~~~~~~~
/mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/ports/esp32/modnetwork_globals.h:38:43: note: in expansion of macro 'MP_ROM_INT'
   38 | { MP_ROM_QSTR(MP_QSTR_AUTH_WPA3_EXT_PSK), MP_ROM_INT(WIFI_AUTH_WPA3_EXT_PSK) },
      |                                           ^~~~~~~~~~
/mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/ports/esp32/modnetwork_globals.h:39:65: error: 'WIFI_AUTH_WPA3_EXT_PSK_MIXED_MODE' undeclared here (not in a function); did you mean 'MP_QSTR_AUTH_WPA3_EXT_PSK_MIXED_MODE'?
   39 | { MP_ROM_QSTR(MP_QSTR_AUTH_WPA3_EXT_PSK_MIXED_MODE), MP_ROM_INT(WIFI_AUTH_WPA3_EXT_PSK_MIXED_MODE) },
      |                                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/py/obj.h:90:67: note: in definition of macro 'MP_OBJ_NEW_SMALL_INT'
   90 | #define MP_OBJ_NEW_SMALL_INT(small_int) ((mp_obj_t)((((mp_uint_t)(small_int)) << 1) | 1))
      |                                                                   ^~~~~~~~~
/mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/ports/esp32/modnetwork_globals.h:39:54: note: in expansion of macro 'MP_ROM_INT'
   39 | { MP_ROM_QSTR(MP_QSTR_AUTH_WPA3_EXT_PSK_MIXED_MODE), MP_ROM_INT(WIFI_AUTH_WPA3_EXT_PSK_MIXED_MODE) },
      |                                                      ^~~~~~~~~~
[14/90] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3...ectronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/extmod/modre.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/ports/esp32/build-GENERIC_S3_N16R8/log/idf_py_stderr_output_46877 and /mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/ports/esp32/build-GENERIC_S3_N16R8/log/idf_py_stdout_output_46877
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [Makefile:66: all] Error 1
`

Any idea what might be causing this?

robert-hh, 2024-04-23:
Which version of the esp-idf do you use?

> Currently MicroPython supports v5.0.4, v5.0.5, v5.1.2, v5.2.0

AppsByDavideV, 2024-04-23:
Hi jasmohan-narula,
I have MPY 1.22 already compiled for that chip, including extra libraries for web server. 
For who's interested, it is here:
[https://drive.google.com/drive/folders/1yNhzdqYlssYW38thWPQJVwSy_E4qJJuO?usp=sharing](url)

There are also some other chip in the folder
Cheers

jasmohan-narula, 2024-04-23:
> Hi jasmohan-narula, I have MPY 1.22 already compiled for that chip, including extra libraries for web server. For who's interested, it is here: [https://drive.google.com/drive/folders/1yNhzdqYlssYW38thWPQJVwSy_E4qJJuO?usp=sharing](url)
> 
> There are also some other chip in the folder Cheers

@AppsByDavideV Thanks a lot. I'll try your binary as well.
I also want to learn to build the binary at my end, so that I can build future versions of micropython as well.

AppsByDavideV, 2024-04-23:
I can understand @jasmohan-narula , that's what I did, so I can keep it updated .

jasmohan-narula, 2024-04-23:
> Hi jasmohan-narula, I have MPY 1.22 already compiled for that chip, including extra libraries for web server. For who's interested, it is here: [https://drive.google.com/drive/folders/1yNhzdqYlssYW38thWPQJVwSy_E4qJJuO?usp=sharing](url)
> 
> There are also some other chip in the folder Cheers

@AppsByDavideV I tried your binary, it's working for me.

jasmohan-narula, 2024-04-23:
After changing the espressif IDE to v5.2

```
$ idf.py --version
ESP-IDF v5.2.1-393-gea967cf2fb
```

I tried building the image. This time I got a new error. So progress.
```
Executing "ninja all"...
[1/1] cd /mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firm...are/micropython/ports/esp32/build-GENERIC_S3_N16R8/bootloader/bootloader.bin
Bootloader binary size 0x4ad0 bytes. 0x3530 bytes (42%) free.
[6/31] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/uart.c.obj
FAILED: esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/uart.c.obj

/mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/ports/esp32/uart.c:60:34: warning: passing argument 2 of 'uart_hal_get_sclk' from incompatible pointer type [-Wincompatible-pointer-types]
   60 |     uart_hal_get_sclk(&repl_hal, &sclk); // To restore SCLK after uart_hal_init() resets it
      |                                  ^~~~~
      |                                  |
      |                                  uart_sclk_t * {aka soc_periph_uart_clk_src_legacy_t *}
In file included from /mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/ports/esp32/uart.c:38:
/mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/esp/esp-idf/components/hal/include/hal/uart_hal.h:209:67: note: expected 'soc_module_clk_t *' but argument is of type 'uart_sclk_t *' {aka 'soc_periph_uart_clk_src_legacy_t *'}
  209 | void uart_hal_get_sclk(uart_hal_context_t *hal, soc_module_clk_t *sclk);
      |                                                 ~~~~~~~~~~~~~~~~~~^~~~
In file included from /mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/esp/esp-idf/components/esp_system/include/esp_system.h:12,
                 from /mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/ports/esp32/mpconfigport.h:10,
                 from /mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/py/mpconfig.h:91,
                 from /mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/py/mpstate.h:31,
                 from /mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/py/runtime.h:29,
                 from /mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/ports/esp32/uart.c:30:
/mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/ports/esp32/uart.c:70:24: error: 'uart_periph_signal' undeclared (first use in this function)
   70 |         esp_intr_alloc(uart_periph_signal[MICROPY_HW_UART_REPL].irq,
      |                        ^~~~~~~~~~~~~~~~~~
/mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/esp/esp-idf/components/esp_common/include/esp_err.h:105:30: note: in definition of macro 'ESP_ERROR_CHECK'
  105 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
/mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/ports/esp32/uart.c:70:24: note: each undeclared identifier is reported only once for each function it appears in
   70 |         esp_intr_alloc(uart_periph_signal[MICROPY_HW_UART_REPL].irq,
      |                        ^~~~~~~~~~~~~~~~~~
/mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/esp/esp-idf/components/esp_common/include/esp_err.h:105:30: note: in definition of macro 'ESP_ERROR_CHECK'
  105 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
[15/31] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/esp32_partition.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/ports/esp32/build-GENERIC_S3_N16R8/log/idf_py_stderr_output_24631 and /mnt/d/3D-Modelling-and-Electronics/My-Projects/ESP32-S3-N16R8-Firmware/micropython/ports/esp32/build-GENERIC_S3_N16R8/log/idf_py_stdout_output_24631
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [Makefile:66: all] Error 1
```

AppsByDavideV, 2024-04-24:
@jasmohan-narula , in the link above there is also a text file with some directions step by step, under Windows WSL. It is in italian, you can easily translate it.

jonnor, 2024-08-24:
GENERIC_S3_SPIRAM_OCT images are now published on MicroPython download page. So there should be no need to build anything custom to run on these devices.

closed 2024-08-27 by projectgus