micropython/micropython #11056: Can't build rp2 Port when MICROPY_HW_ENABLE_USBDEV is disabled

Sebi43217, 2023-03-17:
Im building a projekt for the rp2 with no usb funcionality.
* Im unable to Build the port once i disable  MICROPY_HW_ENABLE_USBDEV in the mpconfigport.h

Is mikrophon witout usb not suppoeted?

[image]

robert-hh, 2023-03-17:
It looks as if tinyusb must not be included in that case. Something to change in the make files. Why do you want to disable USB that way? Would it be sufficient to set `MICROPY_HW_USB_CDC (0)`?

robert-hh, 2023-03-17:
So I have a branch now which allows to unset MICROPY_HW_ENABLE_USBDEV https://github.com/robert-hh/micropython/tree/rp2_usb_deinit
Three files are changed, and the setting for MICROPY_HW_ENABLE_USBDEV must be done in CMakeFiles.txt.

robert-hh, 2023-03-18:
I changed that branch to have all modifications in .h and .c files. That's style-wise more consistent. No change in the Make files.

dpgeorge, 2023-03-20:
Fixed by be686e634eede19f33612122d7254007a3d72bd0

closed 2023-03-20 by dpgeorge

Sebi43217, 2023-03-20:
Thanks for the quick fix it works perfectly now.