micropython/micropython #11057: REPL over UART not working on the RP2 port

Sebi43217, 2023-03-17:
I’m trying to run a clean micropython port for the rp2 using a picoprobe as a debugger but I’m unable to use the REPL over UART functionality. 

* The only modification from a clean micropiton checkout is that i enabled  MICROPY_HW_ENABLE_UART_REPL in the mpconfigport.h

* The REPL over UART sems to immediately crash after startup since its not responding after it has send its startup information.

* When using rshell its getting stuck at "Retrieving sysname ..."
[image]

* Using a terminal like minicom I can see the startup message but I’m unable to get any future response from the rp2.
[image]

robert-hh, 2023-03-17:
I can enable UART REPL and it works fine, with and without USB CDC enabled. Tested with a simple USB/UART adapter.

robert-hh, 2023-03-17:
I just noticed that UART REPL input only works if DEBUG is disabled. For builds with DEBUG=1 uart REPL does not take input values.

robert-hh, 2023-03-17:
UART0 seems to be disabled by DEBUG code in the SDK. But there is a hack in the code to re-enable it.
Add to main.py the lines:
```
from machine import freq
freq(125000000)
```
The actual freq value does not matter, but the default is fine.

Sebi43217, 2023-03-18:
Thanks that could be the solution since I was always working with DEBUG enabled.

Sadly I'm away for the weekend but I'll try it out on Monday.

robert-hh, 2023-03-18:
I made a branch with a suitable change. There is one line changed in main.c, line 93-95, to:
```
    #if !defined(NDEBUG) && !MICROPY_HW_ENABLE_UART_REPL
    stdio_init_all();
    #endif
```
Actually, only line 93 is changed. And I removed the patch in modmachine.c in the code for machine.freq().
The branch is here: https://github.com/robert-hh/micropython/tree/rp2_keep_uart0

dpgeorge, 2023-03-20:
Fixed by 711bac511e547096fe1e58af3ba70d31b244731f

closed 2023-03-20 by dpgeorge