micropython/micropython #11061: tell() always return 0 on RP2 port when appending

ThinkTransit, 2023-03-18:
When calling the tell() method on existing files 0 is always returned.

```python
MicroPython v1.19.1-921-gfb94ae48e-dirty on 2023-03-07; P4 with RP2040
Type "help()" for more information.
>>> f = open('main.py', 'a')
>>> f.tell()
0
>>> 
```

Expected behaviour (cpython)
```python
Python 3.10.7 (main, Nov 24 2022, 19:45:47) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> f = open('main.py', 'a')
>>> f.tell()
5059
>>> 
```

robert-hh, 2023-03-18:
I can confirm the behavior. Not surprisingly, it is not limited to the rp2040 port. 
After writing at least one character, the proper position is reported.
Obviously the same happens, if a f.seek(0, 2) is done (position to the end).

ThinkTransit, 2023-03-18:
Thanks @robert-hh 

I wonder if this could be a littlefs issue maybe?

robert-hh, 2023-03-19:
It seems indeed to be a LFS issue, for both LFS1 and LFS2. With FAT it looks fine.