micropython/micropython #11067: Socket accept connections every other time

lukasz-blonski, 2023-03-18:
UF2 used [v1.19.1-966-g05bb26010 (2023-03-13) .uf2](https://micropython.org/resources/firmware/rp2-pico-w-20230313-unstable-v1.19.1-966-g05bb26010.uf2)

Board: Pico W

Running a simple web server, connected to WiFi works every other time. A simple step to reproduce is to:
1) power up the device, if it accepts the connection, unplug the USB cable
2) pico w will no longer accept any connect
[scratch_2.patch](https://github.com/micropython/micropython/files/11009586/scratch_2.patch)
ions, unplug the USB cable
3) pico w will now accept new connections, unplug the USB cable
4) go to 2)

I tried on 2 pico w boards both using the blocking and nonblocking sockets and using the uasyncio start_server approach. The behaviour is consistent, the board will accept the connection only every other time when powered up. When debugging with the socket based code, I noticed that the program times out on `socket.accept` call and it never returns (or throws).

Attaching MicroPython code I used to test it with (one of many variants that never worked)

dpgeorge, 2023-03-20:
Thanks for the report.

I cannot follow exactly what the steps are above to reproduce the issue.  In every step it says to "unplug the USB cable", but no steps tell to plug it back in.  Also how is the board powered if the USB cable is not plugged in?  These details are important.

The code in scratch_2.patch does not run as-is.  It needs some imports, and the code at the end must be wrapped in an async function.  Also it needs an `index.html` file in the filesystem.  After making these changes I was able to run that code on a Pico W and it works OK for me, it allows connections to be made one after the other and they all work.

Can you please describe the exact steps to reproduce the issue you are seeing?

lukasz-blonski, 2023-03-20:
Thanks for the reply and apologies for the incomplete code, I tried multiple variations of this simple server implementation and the result would be the same.

This code also works for me but only 50% of the time.

Basically, it stops working after I do a hard restart of Pico W by unplugging the USB cable. Then when I do a hard restart again (unplug/ plug the cable) it works. Hard restart - it doesn't work, another hard restart it works.

I was able to reproduce this behaviour on 2 separate boards with multiple variations of the patch (with and without uasyncio).

Hope this helps.

dpgeorge, 2023-03-20:
> Basically, it stops working after I do a hard restart of Pico W by unplugging the USB cable.

In order to make progress on this you'll need to provide more information.  You need to provide exact steps and code so someone else can follow it and reproduce it.

For example:
- when and how do you connect to the WiFi after doing the hard reset?
- when and how do you start the web server?
- what browser (or otherwise) do you use to access the rp2 web server?
- how do you know the IP address of the rp2?
- give a timeline, eg 5 seconds after starting the web server the browser accesses the rp2

I have not seen this kind of issue before and it's most likely related to your set up, eg router, WiFi network.

lukasz-blonski, 2023-03-20:
Very well - I will prepare a fully working script along with the code to connect to wifi and answer all the questions to help with the reproduction of this issue. Appreciate the help!