micropython/micropython #11078: Any plans to build on ESP-IDF v5.0 for the ESP32 port?

ma261065, 2023-03-20:
I believe there are a number of breaking changes in ESP-IDF v5.0, but as v4.4 is already in the maintenance period, and will be unsupported in Jul 2024, at some point we will have to move off it.

Also, some of the newer chips (C2 & C6) are only supported in v5.0

Wondering if the changes are massive, and whether it is worth starting the work now (maybe for the 1.20 release) rather than waiting?

cherrydev, 2023-03-29:
I would like to see this happen as well. I have a specific use-case I'd like to make use of: ESP-IDF 5.x supports DMA transfers on the RMT peripheral for some of the newer ESP chips. Based on the tests I've done with the raw IDF, I'd be able to contribute a WS2812b component that would be able to drive thousands of LEDs simultaneously over multiple channels. The current WS2812b component is 100% synchronous and blocks the main thread for the entire data transmission.

From what I've seen so far there are some IDF 4.x components that were deprecated and are now removed from 5.x. Some of the components still exist but have been moved to an external component repository. If the micropython team expresses an intention to begin the migration process I would like to contribute to the migration.

jkhax0r, 2023-03-31:
FWIW, I saw there was a branch/fork/PR for IDF 5.0 over on CircuitPython that may be helpful.  Not sure what the status was or anything

cherrydev, 2023-04-01:
Huh, okay. Looks like they've gotten a fair distance on it. But isn't circuitpython supposedly downstream from micropython? Or is that not the case anymore?

dhalbert, 2023-04-01:
> But isn't circuitpython supposedly downstream from micropython?

CircuitPython uses the core Python language implementation from MicroPython, and some of the port-independent code (`re`, etc.). But the built-in modules are mostly different.

dpgeorge, 2023-04-06:
There is a PR at #8777 which has started work on supporting IDF v5.  We definitely want to support that version of the IDF, so feel free @cherrydev to help get that PR in a state where it can be merged.

cherrydev, 2023-04-06:
@dpgeorge that's great! I'll take a look.

Carglglz, 2023-04-06:
@cherrydev be aware of this too  #8988 👍🏼 it will be needed to update to IDF v.5

jorisplusplus, 2023-05-10:
I have gotten a build to work on the esp-idf v5.0 based on current master branch with the changes listed in #8988. There is currently still a lot broken most notably the network stack. But the it boots and the REPL works with some basic commands.
I have noticed that there are quite some significant changes between V5.0 and V4.x.
Is the goal to keep support for V4.x, or to drop v4.x and move to v5.0 only?

dpgeorge, 2023-05-11:
> I have gotten a build to work on the esp-idf v5.0 based on current master branch

I also managed to do this recently (with help from @projectgus for the UART REPL).  My branch is at https://github.com/dpgeorge/micropython/tree/esp32-idf-50 but I didn't yet make a PR.

> Is the goal to keep support for V4.x, or to drop v4.x and move to v5.0 only?

I propose to drop support for IDF v4.x completely, and move exclusively to v5.0.1 (and later to higher v5 versions when they are released).

jorisplusplus, 2023-05-12:
> I also managed to do this recently (with help from @projectgus for the UART REPL). My branch is at https://github.com/dpgeorge/micropython/tree/esp32-idf-50 but I didn't yet make a PR.

That is looking a lot better than what I have currently. I'm willing to help if necessary.

dpgeorge, 2023-05-17:
See #11528 for my IDF 5 branch/PR.

ma261065, 2023-05-20:
It seems that even Expressif is still using v4.4 for their Arduino Core. Just linking to their issue to track the update to v5.x in case there are any useful learnings there - https://github.com/espressif/arduino-esp32/issues/7852

dpgeorge, 2023-05-22:
> It seems that even Expressif is still using v4.4 for their Arduino Core. Just linking to their issue to track the update to v5.x in case there are any useful learnings there

Thanks, that's useful to know about and read.  It seems Arduino Core is being updated straight to IDF v5.1, instead of v5.0 which is what we will do here (there is currently no v5.1 release but there is a v5.0 release).

And it seems they are stopping work on their IDF v4.4 support and only doing bug fixes for that.

So I think this is a good indication that we are doing the right thing updating to v5.0.

andrewleech, 2023-08-10:
@ma261065 IDF5 support has now merged to master in https://github.com/micropython/micropython/pull/11528 ! Is there any other outstanding issues you're aware of? If not this issue can probably be closed?

ma261065, 2023-08-10:
Closing as completed

closed 2023-08-10 by ma261065