micropython/micropython #11083: Can't find esp idf header file

MATTYGILO, 2023-03-20:
I am trying to change the ble channel of my device, I am creating a custom user module but I get this error:

I am building with 
esp idf commit:
5bb59b00e72f8f91eb24d8c65bf9a7ea2b8a4f5f

Micropython:
latest

```
/ble/ble.c:4:10: fatal error: esp_gap_ble_api.h: No such file or directory
 #include "esp_gap_ble_api.h"
```

This is my usercmodule code:

```
#include <stdio.h>
#include "py/nlr.h"
#include "py/runtime.h"
#include "esp_gap_ble_api.h"

STATIC mp_obj_t ble_set_channel_mask(mp_obj_t mask_obj) {
    esp_ble_adv_channel_t mask = mp_obj_get_int(mask_obj);
    esp_ble_gap_set_channel_mask(mask);
    printf("BLE channel mask set to %d\n", mask);
    return mp_const_none;
}

STATIC MP_DEFINE_CONST_FUN_OBJ_1(ble_set_channel_mask_obj, ble_set_channel_mask);

STATIC const mp_map_elem_t ble_module_globals_table[] = {
    { MP_OBJ_NEW_QSTR(MP_QSTR___name__), MP_OBJ_NEW_QSTR(MP_QSTR_ble) },
    { MP_OBJ_NEW_QSTR(MP_QSTR_set_channel_mask), (mp_obj_t)&ble_set_channel_mask_obj },
};

STATIC MP_DEFINE_CONST_DICT(ble_module_globals, ble_module_globals_table);

const mp_obj_module_t ble_module = {
    .base = { &mp_type_module },
    .globals = (mp_obj_dict_t*)&ble_module_globals,
};
```

This is my micropython.cmake:

```
add_library(usermod_ble INTERFACE)

target_sources(usermod_ble INTERFACE
    ${CMAKE_CURRENT_LIST_DIR}/ble.c
)

target_include_directories(usermod_ble INTERFACE
    ${CMAKE_CURRENT_LIST_DIR}
)

target_link_libraries(usermod INTERFACE usermod_ble)
```

This is my micropython.mk:

```
MBEDTLS_MOD_DIR := $(USERMOD_DIR)

SRC_USERMOD += $(MBEDTLS_MOD_DIR)/ble.c
```

robert-hh, 2023-03-20:
Checking with esp-idf v4.2.2 and 4.4.2, the file exists at esp-idf/components/bt/host/bluedroid/api/include/api. So it might require to adapt the include path or the path name for the include.

MATTYGILO, 2023-03-20:
What alternative paths do you think it could be?

robert-hh, 2023-03-20:
Since esp-idf seem to be in the include path, you could try to include:
#include "components/bt/host/bluedroid/api/include/api/esp_gap_ble_api.h"
Or you keep the include and try to add in CMakeLists.txt the path to target_include_directories (line 202 ..), e.g.:
```
target_include_directories(${MICROPY_TARGET} PUBLIC
    ${IDF_PATH}/components/bt/host/nimble/nimble
    ${IDF_PATH}/components/bt/host/bluedroid/api/include/api
)
```

MATTYGILO, 2023-03-20:
Thanks for the suggestions @robert-hh so I've gone about testing them

Suggestion 1: #include "components/bt/host/bluedroid/api/include/api/esp_gap_ble_api.h"
Suggestion 2: target_include_directories

Suggestion 1 result:
```
/ble/ble.c:5:10: fatal error: components/bt/host/bluedroid/api/include/api/esp_gap_ble_api.h: No such file or directory
 #include "components/bt/host/bluedroid/api/include/api/esp_gap_ble_api.h"
```

Suggestion 2 result (using esp_gap_ble_api.h):
```
FAILED: esp-idf/main/CMakeFiles/BUILD_FROZEN_CONTENT frozen_content.c micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/esp-idf/main/CMakeFiles/BUILD_FROZEN_CONTENT micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/frozen_content.c 

MPY apa106.py
error compiling apa106.py:
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

Suggestion 1 + 2 result:
```
/ble/ble.c:5:10: fatal error: components/bt/host/bluedroid/api/include/api/esp_gap_ble_api.h: No such file or directory
 #include "components/bt/host/bluedroid/api/include/api/esp_gap_ble_api.h"
```

robert-hh, 2023-03-20:
Which version of the esp-idf do you use? You showed the git hash, bit not the tag. Please run in esp-idf the command:
git branch

For instance, if I do that, I get:
```
esp-idf$ git branch 
* (HEAD detached at v4.4.2)
  master
  release/v4.3
```

For option 2, the #include statement should be :` #include "esp_gap_ble_api.h"`, because the remainder of the path is defined in the target_include_directories setting.

MATTYGILO, 2023-03-20:
Ok so I used `` and ``. This is my full error:

```
[10/245] Generating ../../genhdr/root_pointers.collected
Root pointer registrations not updated
[11/244] Generating ../../genhdr/qstrdefs.collected.h
QSTR not updated
[12/242] cd /micropython/...s//micropython/ports/esp32/boards/manifest.py
FAILED: esp-idf/main/CMakeFiles/BUILD_FROZEN_CONTENT frozen_content.c /micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/esp-idf/main/CMakeFiles/BUILD_FROZEN_CONTENT /micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/frozen_content.c
cd /micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/esp-idf/main && /venv/bin/python3.10 /micropython/tools/makemanifest.py -o /micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/frozen_content.c -v MPY_DIR=/micropython -v MPY_LIB_DIR=/micropython/lib/micropython-lib -v PORT_DIR=/micropython/ports/esp32 -v BOARD_DIR=/micropython/ports/esp32/boards/GENERIC_S3_SPIRAM_OCT -b /micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT -f-march=xtensawin --mpy-tool-flags= /micropython/ports/esp32/boards/manifest.py
MPY apa106.py
error compiling apa106.py:

ninja: build stopped: subcommand failed.
ninja failed with exit code 1
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [Makefile:52: all] Error 1
```

robert-hh, 2023-03-20:
The esp-idf version seems ok. And the error you get seems not related to the C code you want to compile. At least there is not complaint about the header file not found. Did you do a `make clean` before compiling to have a full fresh start?

MATTYGILO, 2023-03-20:
> Ok just ran `make BOARD=GENERIC_S3_SPIRAM_OCT clean` and ran the build again still the same problem as before

For accuracy it is now crashing at [1171/1401]
```
[1171/1401] cd /micropyth...s//micropython/ports/esp32/boards/manifest.py
FAILED: esp-idf/main/CMakeFiles/BUILD_FROZEN_CONTENT frozen_content.c /micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/esp-idf/main/CMakeFiles/BUILD_FROZEN_CONTENT /micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/frozen_content.c 
cd /micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/esp-idf/main && /venv/bin/python3.10 /micropython/tools/makemanifest.py -o /micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/frozen_content.c -v MPY_DIR=/micropython -v MPY_LIB_DIR=/micropython/lib/micropython-lib -v PORT_DIR=/micropython/ports/esp32 -v BOARD_DIR=/micropython/ports/esp32/boards/GENERIC_S3_SPIRAM_OCT -b /micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT -f-march=xtensawin --mpy-tool-flags= /micropython/ports/esp32/boards/manifest.py
MPY apa106.py
error compiling apa106.py:

ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

robert-hh, 2023-03-20:
The error message is not very informative. Do you get better info if you run with the option V=1. That creates lots of messages, but at least it may show the actual compile errors for frozen_content.c