micropython/micropython #11085: RP2 include example_user_cmodule not working

Sebi43217, 2023-03-20:
I’m trying to write my own c-module and started with the Basic example from the Documentation under:

https://docs.micropython.org/en/latest/develop/cmodules.html

I’m running "make USER_C_MODULES=../../examples/usercmodule/micropython.cmake" in the rp2 folder to include the Module.
In the build output it sems to find and include the module:
[image]

But I cant access the module in micropython:
[image]

Are there any extra steps required for the RP2 port?

dpgeorge, 2023-03-21:
I tried the above "make" command and it worked OK for me.  I could import both `cexample` and `cppexample`.

Make sure you do a "make clean" first (remove the build directory).

Sebi43217, 2023-03-21:
Thank you for the evaluation of my problem.

I am working with a vscode setup according to the getting started from raspberry and use a picoprobe to debug and flash my target. 

I have now found the issue to be that the supplied launch.json from raspberry builds the project again and which has to be disabled in the settings.json.

The second issue is that it builds and searches the binary in the folder build and micropython builds under build-PICO.

To evaluate I have copied the contents of build-Pico into build and then it runs fine.

I’m right now trying to find out how I have to modify the launch.json to use the build-PICO folder.

Sebi43217, 2023-03-21:
I have fixed it by adding ""cmake.buildDirectory": "${workspaceFolder}/build-PICO"," to the settings.json and restarting vscode / running configure.

This is probably not the optimal way but works fine for me.