micropython/micropython #11101: pyboard.py does not keep directory structure like proposed by docs

paulober, 2023-03-22:
- Current pyboard.py tool from the master branch

When i upload files like described in following docs:
```py
# Copy three files to the device, keeping their names
# and paths (note: `lib` must exist on the device)
$ pyboard.py --device /dev/ttyACM0 -f cp main.py app.py lib/foo.py :
```
It puts all files into the root of the target device even if lib exists on the target device.

JohannesMaierhofer, 2024-04-29:
same happens for the generic ESP32S3 version by uploading via the Thonny interface.

jonnor, 2024-09-01:
Note that we now also have a `mpremote` tool, which can do many of the same things. The pyboard.py docs are however still there, so issue still seems relevant.

dpgeorge, 2024-09-02:
I tested this and the original report is correct, the docs don't match the behaviour.

Given #11777 will be merged soon, I suggest to just change the docs to match the current behaviour, which doesn't preserve directory names when copying to the device.

dpgeorge, 2024-09-02:
See #15758 for a docs fix for this issue.

dpgeorge, 2024-09-17:
Fixed by 3d53b39a2a8b0d49f93a6c5915b22caa9ddae6d8

closed 2024-09-17 by dpgeorge