micropython/micropython #11116: rp2: Reserve RAM for C++ heap

Gadgetoid, 2023-03-24:
This commit assumes that all modules are C-only modules, with fairly explicit control over malloc/free - https://github.com/micropython/micropython/commit/c80e7c14e6305e50e3b39f97172d4d8fe1214d3b

That's fair enough, but - for better or worse - MicroPython explicitly supports C++ modules where such guarantees appear to be absent.

*The TLDR is that - I think - *any* C++ modules included in MicroPython may trigger heap allocation via new/delete that the author doesn't have full control over. With the changes linked above, this causes the C++ heap and MicroPython's GC heap to overlap and contest the same space, unknowingly overwriting each other and causing hardlocks and weird, unpredictable errors.*

I have spent two weeks prodding and poking our C++ codebase, removing explicit uses of `new` and `delete` in favour of deferring to MicroPython's `m_new`, which we use to allocate C++ classes with "placement new", eg:

```c++
void *mem = m_new(Cls, 1);
Cls *instance = new(mem) Cls();
```

Which allows us to allocate C++ classes on the MicroPython heap. Since these are very closely bound to MicroPython classes, we can store a reference such that they are not garbage collected.

Try as I might, however, I have been unable to trace and clean up *implicit* uses of `new` and `delete` that pertain to C++ features. This could include things like lambda functions which allocate on heap when taking a copy of a captured variable, or even just casual use of `std::string` (though the latter is a bad example since `string_view` works well with MicroPython's data/len pairs for string data.)

I created a C++ module that would track our heap usage, and found that *just* starting up MicroPython with our modules compiled in would use some 9,626 bytes over 400 allocations. By editing `memmap_mp.ld` to avoid the first 10k RAM, I was able to get a stable (at least until a large C++ heap allocation happens) build of MicroPython:

```patch
diff --git a/ports/rp2/memmap_mp.ld b/ports/rp2/memmap_mp.ld
index 6be05b094..5e61dd4c5 100644
--- a/ports/rp2/memmap_mp.ld
+++ b/ports/rp2/memmap_mp.ld
@@ -249,7 +249,7 @@ SECTIONS
     __StackOneBottom = __StackOneTop - SIZEOF(.stack1_dummy);
     __StackBottom = __StackTop - SIZEOF(.stack_dummy);
     /* Define start and end of GC heap */
-    __GcHeapStart = __bss_end__;
+    __GcHeapStart = __bss_end__ + (10 * 1024);
     __GcHeapEnd   = __StackLimit;
     PROVIDE(__stack = __StackTop);
```

Off the back of this, I'd like to request some consideration about C++ and how it might interact with this change. I don't claim to have any particularly good answers since having inscrutable memory allocation that the MicroPython user can't see is... not great... but there are a some options:

### The "just gimme some C++ RAM" method

Implement a per-board configuration, compile switch or other machinery to allow a region of RAM to be reserved for C++.

The linker script should also more carefully reserve the `GcHeap` region, such that it's not considered "free" by C++. This would make heap allocations fail fast and more predictably with a hardfault, rather than corrupting user code and (potentially) data.

### The "make new/delete work in terms of GC heap" method

It feels like it should be possible to implement `new` and `delete` such that they allocate on MicroPython's heap, taking into account that the heap would need to be ready *very early* in program execution... somehow.

I gave it a try, but since C++ objects can be initialised very early in `runtime_init`: https://github.com/raspberrypi/pico-sdk/blob/f396d05f8252d4670d4ea05c8b7ac938ef0cd381/src/rp2_common/pico_runtime/runtime.c#L104-L106, before MicroPython's heap is initialised. I'm afraid this is quite beyond me- my attempts to get it working, even by brute force, have met with failure.

There are benefits and drawbacks to this solution. Since MicroPython uses mark-and-sweep we'd need to somehow flag long-lived (survive soft reset) C++ objects so that C++ isn't trashed upon a soft reset. Additionally having C++ allocate but not free long-lived objects mid-way through the GC heap would cause fragmentation, which is BAD.

### The "roll our own allocator" method

It's not outside the realm of possibility to just create a new C++ module that a user could link against MicroPython and that:

1. Replaces `new` and `delete` with its own implementations
2. Reserves RAM such that it's in .bss and not touched by MicroPython's GC heap
3. Cooperates with MicroPython so that exceptions can be raised when allocations fail

This takes us back to a "split" approach, but since the amount of RAM required by C++ is very particular to an individual use-case, perhaps this is the sensible answer. This approach also has the benefit of being completely separate to MicroPython so you get to pretend that the problem doesn't exist :laughing: 

Thank you!

Gadgetoid, 2023-03-24:
Accepting that I am incredibly out of my depth here, here's an attempt at an allocator module that replaces `new`/`delete` as proposed: https://github.com/Gadgetoid/micropython-cppmem/blob/a8931f40950a3f143c331608f297c57ec3869aad/cppmem.cpp

It does two things:

1. Absorbs all preinit allocations into a 10k linear, non-freeing memory pool that lasts the lifetime of the session (until hard reset)
2. Exposes a method to switch to `MICROPYTHON` mode, whereupon `new` and `delete` map to `m_malloc` and `m_free`

I'm caught in the trap of thinking `m_malloc` might want to be `m_tracked_alloc`, but the intent of remapping C++ allocations is to catch the odd, short-lived straggler and *not* to allow the C++ module author to casually allocate blocks of memory on MicroPython's heap with impunity. This is mostly in line with our requirements and existing practises, rather than a suggested direction.

Right now the switch to MicroPython pooling is manual, but it could theoretically happen automatically just before `main()` using a C++ object instance with a custom init priority.

dpgeorge, 2023-03-29:
> Implement a per-board configuration, compile switch or other machinery to allow a region of RAM to be reserved for C++.
>
> The linker script should also more carefully reserve the `GcHeap` region, [...]

I think we should do this now as (at minimum) a temporary solution, maybe improved in the future.

What linker symbols does the libc malloc (and C++ new) use to find out the extent of its heap?  We can modify `memmap_mp.ld` to make it expose a small region for these symbols, and the remaining is reserved for the MP GC heap.

See also related issue that needs some libc heap: https://github.com/micropython/micropython/pull/10739#issuecomment-1481042783

Gadgetoid, 2023-03-29:
Ah it looks like we do have very comparable problems.

Insofar as I'm aware C++'s `new` uses `sbrk` as provided by the Pico SDK runtime here: https://github.com/raspberrypi/pico-sdk/blob/1.5.0/src/rp2_common/pico_runtime/runtime.c#L191-L216

Internally that uses the `end` symbol as its starting point. This is right after the last global variable. It uses `heap_end` to keep track of where the heap has grown to, and checks for an overlap into `__StackLimit` and fails to grow the heap if it might overlap the stack.

Since MicroPython's heap on RP2 is now allocated with a couple of symbols that greedily take up all RAM, these two concepts overlap. I think we'd have to provide a custom implementation of `sbrk` (easy? because it's marked weak in Pico SDK) that- instead of using `end` and `__StackLimit`- uses its own custom start and end symbols.

It strikes me that it would be preferable to replace `malloc` and `free` with implementations that use MicroPython's heap. I think these can just be defined *somewhere* and the libc implementations will be ignored.

In my case I can just use a module that implements `new` and `delete` in terms of MicroPython heap. This works because all of our memory use is confined to the scope of a single function call, so we can gently tip-toe around GC issues :grimacing: 

lwip *also* has its own heap implementation (https://github.com/lwip-tcpip/lwip/blob/6ca936f6b588cee702c638eee75c2436e6cf75de/src/core/mem.c), which means we've potentially got C++, C and arbitrary other usages contending with MicroPython for the same space and if the user doesn't use any of these features, they've lost that RAM.

I might be wrong, but I don't think C has any runtime init, or at least any that dynamically allocates RAM? The big problem with C++ is casually creating global class instances, or things like `std::map` which are then initialised here (before `main`):

https://github.com/raspberrypi/pico-sdk/blob/1.5.0/src/rp2_common/pico_runtime/runtime.c#L96-L106

In all cases it would be nice to have MicroPython's heap initialised *as early as possible* so that all RAM usage can be trivially redirected there- assuming such liberal use of `m_tracked_alloc` doesn't pose a problem. (Am I right in thinking it's a doubly linked list growing from a single root pointer?)

Gadgetoid, 2023-03-29:
For runtime (not init) C++ code, overloading `sbrk` does exactly what you'd expect. It's called on the first allocation (when the heap needs to grow from 0 to something) and then subsequently for expansion. Just shimming it up like so will make C++ `new` allocate on the MicroPython heap:

```c
void *_sbrk(int incr) {
    mp_printf(&mp_plat_print, "sbrk: %d", incr);

    return m_tracked_calloc(1, incr);
}
```

This approach is flawed, since the heap is expected to grow as a contiguous whole and this will return fragmented blocks. But that's an implementation detail for- I guess - `m_tracked_realloc` :laughing: 

It's also flawed because C++ can try to allocate memory *before* the MicroPython heap is even initialised. Careful authoring of modules can avoid this, but it would be nice to wholly support C++ features without weird caveats.

For C it's much trickier. You can `__wrap_` C functions such as malloc and free, and indeed Pico SDK's runtime does this to make them thread-safe. We would need to mind thread safety and include a mutex, but a very naive implementation like the following works:

```c
// Include MicroPython API.
#include "py/runtime.h"

void *__wrap_calloc(size_t count, size_t size) {
    mp_printf(&mp_plat_print, "calloc: %u", size * count);
    void* result = m_tracked_calloc(count, size);
    if(result == NULL) {
        mp_raise_msg(&mp_type_RuntimeError, MP_ERROR_TEXT("calloc failed"));
    }
    return result;
};

void *__wrap_malloc(size_t size) {
    mp_printf(&mp_plat_print, "malloc: %u", size);
    void* result = m_tracked_calloc(1, size);
    if(result == NULL) {
        mp_raise_msg(&mp_type_RuntimeError, MP_ERROR_TEXT("malloc failed"));
    }
    return result;
};

void *__wrap_realloc(void *mem, size_t size) {
    return NULL;
};

void __wrap_free(void *mem) {
    mp_printf(&mp_plat_print, "free: %lu\n", (uint32_t)mem);
    m_tracked_free(mem);
};
```

Because we're replacing the wrapped functions that Pico SDK's runtime provides, the `pico_malloc` and `pico_malloc_headers` libraries need quietly discarding. This can be achieved in `CMakeLists.txt` by adding our own before the call to `pico_sdk_init()` (yes I know this is a horrible approach and we'll need some upstream support on this):

```cmake
add_library(pico_malloc INTERFACE)
add_library(pico_malloc_headers INTERFACE)
pico_sdk_init()
```

Because Pico SDK *only* creates targets if they don't already exist, this prevents their creation entirely.

Then we must include `malloc.c` with the above C code into the build, and set up the wrapping:

```cmake
set(MICROPY_SOURCE_PORT
    malloc.c
    ...
)

target_link_options(${MICROPY_TARGET} PRIVATE "LINKER:--wrap=malloc")
target_link_options(${MICROPY_TARGET} PRIVATE "LINKER:--wrap=calloc")
target_link_options(${MICROPY_TARGET} PRIVATE "LINKER:--wrap=realloc")
target_link_options(${MICROPY_TARGET} PRIVATE "LINKER:--wrap=free")
```

After fighting with compiler to get the basic C example to *actually allocate some RAM*, I end up with regular C malloc/free redirected via my own wrappers onto the MicroPython heap:

```python
>>> import cexample
>>> cexample.add_ints(1, 1024)
malloc: 4096
ptr: 536920680
free: 536920680
1025
```

Modifications to the example code:

```c
#include <stdlib.h>

// This is the function which will be called from Python as cexample.add_ints(a, b).
STATIC mp_obj_t example_add_ints(mp_obj_t a_obj, mp_obj_t b_obj) {
    // Extract the ints from the micropython input objects.

    int *buf = malloc(1024 * sizeof(int));
    buf[0] = mp_obj_get_int(a_obj);
    buf[1] = mp_obj_get_int(b_obj);

    mp_printf(&mp_plat_print, "ptr: %lu\n", (uint32_t)buf);

    // Calculate the addition and convert to MicroPython object.
    mp_obj_t result = mp_obj_new_int(buf[0] + buf[1]);
    free(buf);
    return result;
}
```

This ... approach ... would, I think, fix the Bluetooth problem without needing to monkey around providing module-specific implementations.

*Unfortunately* it may not be widely portable to other platforms which may potentially do their own monkeying with `malloc` and co.

Gadgetoid, 2023-03-29:
I am pretty sure this approach would also remove the fixed 8019 byte `ram_heap` used by lwIP, can just add:

```patch
diff --git a/ports/rp2/CMakeLists.txt b/ports/rp2/CMakeLists.txt
index 838d4271e..35ac75558 100644
--- a/ports/rp2/CMakeLists.txt
+++ b/ports/rp2/CMakeLists.txt
@@ -213,6 +213,7 @@ if (MICROPY_PY_LWIP)
     )
     target_compile_definitions(${MICROPY_TARGET} PRIVATE
         MICROPY_PY_LWIP=1
+       MEM_LIBC_MALLOC=1
     )
 endif()
 

```

But some testing seems to suggest it's *really unstable*, I'm guessing because it does rather a lot, and maybe needs a particular alignment?:

```
Connecting...
malloc: 76
free: 536946352
malloc: 72
free: 536946352
malloc: 244
free: 536971248
malloc: 72
free: 536946720
malloc: 72
free: 536946720
malloc: 72
free: 536946720
malloc: 72
free: 536976528
malloc: 148
malloc: 744
free: 536977936
free: 536981584
malloc: 72
free: 536977936
malloc: 144
malloc: 748
free: 536970352
free: 536970496
malloc: 72
free: 536970352
malloc: 72
free: 536970352
malloc: 72
```

peterharperuk, 2023-03-30:
I was under the impression that we couldn't use MEM_LIBC_MALLOC=1 because we would attempt to use malloc from an IRQ context? lwip and cyw43 is running in a pendsv interrupt context.

Gadgetoid, 2023-03-30:
That would explain why I saw intermittent freezes, though it sort-of worked. I'm thundering into something I don't fully understand here, since the C `malloc`/`free` tinkering was a distraction from the C++ problem I was trying to solve (which I also don't fully understand).

dpgeorge, 2023-06-02:
Commit fe4ac49d7f3beb0512545ac8e4545c76da2334ec allowed a board to reserve some memory for the C heap, by defining `MICROPY_C_HEAP_SIZE`.