micropython/micropython #11122: can´t compile

hdpklm, 2023-03-25:
Hello

os: debian:11-slim in docker containder

I try to compile the code using the instruction in this utl
https://github.com/micropython/micropython/tree/master/ports/esp32
but its not work, can you please do the instruction easy?

when I run this command i get errors

```sh
$ make -C mpy-cross
make: *** mpy-cross: No such file or directory.  Stop.
```

```sh
$ cd ports/esp32
sh: cd: ports/esp32: No such file or directory
```

daed, 2023-04-18:
I found much better results with using the espressif docker images:

```FROM espressif/idf:v4.2 AS base```

I've since switched to Pico W from esp32 (the espressif image also works great for this) but I still have the esp32 work I was doing at https://github.com/bitflange/esp32-micropython-factory, if that helps.

hdpklm, 2023-04-19:
hi @daed 
very interesting
can you please write step by step how to install and run the code
thank you so much