micropython/micropython #11140: Reading pin value from IRQ hangs forver

GuillaumeLeclerc, 2023-03-28:
```python
from machine import Pin, idle, disable_irq, enable_irq
from time import sleep_ms

pin = Pin(1, Pin.IN)

def on_change(pin):
    state = disable_irq()
    pin.value()
    enable_irq(state)

pin.irq(on_change, Pin.IRQ_RISING | Pin.IRQ_FALLING)

for i in range(1000):
        idle()
        sleep_ms(100)
        print("WORKING")

```

It seems impossible to read the value of the pin object passed as an argument to the callback. As soon as it is triggered the RPI Pico stop working until I unplug/replug it. Using the pin from the global context works or even recreating a new `Pin` instance, but not using the one passed as an argument

firmware: `rp2-pico-20230116-unstable-v1.19.1-803-g1583c1f67.uf2`

robert-hh, 2023-03-28:
I tested it here with a slight change, using pin 3 instead of pin 1 for no specific reason, but it works fine. I can even print the value of the pin in the irq function. The enable/disable irq is not required.
Which frequency do you apply to the pin? I used a 1 Hz square from a signal generator. If you use a manual method or a switch, there may be fast bursts due to contact bouncing.
Do you have UART REPL active on Pin 0 and 1. Pin 1 would be the RX pin.

GuillaumeLeclerc, 2023-03-28:
I just have the repl that is forwarded to USB and if I understand properly in the RP2040 there is an internal USB to UART so it shouldn't affect Pin 0 and 1.

The signal I'm feeding is a square wave that runs around 1KHz, filtered by capacitors. Again if I replace `pin` from the arguments by either the `pin` from the global context it works fine

`Pin(1, Pin.Input).value()` also doesn't hang

dpgeorge, 2023-03-29:
The above code works OK for me, using `v1.19.1-992-g38e7b842c`.

I also print the `pin` object and `pin.value()` value in the IRQ handler, and the current idle loop iteration.  Output is like this:
```
WORKING 275
WORKING 276
WORKING 277
WORKING 278
WORKING 279
WORKING 280
WORKING 281
Pin(GPIO1, mode=IN) 1
Pin(GPIO1, mode=IN) 0
Pin(GPIO1, mode=IN) 1
Pin(GPIO1, mode=IN) 0
WORKING 282
Pin(GPIO1, mode=IN) 1
Pin(GPIO1, mode=IN) 0
Pin(GPIO1, mode=IN) 1
Pin(GPIO1, mode=IN) 0
Pin(GPIO1, mode=IN) 1
Pin(GPIO1, mode=IN) 0
Pin(GPIO1, mode=IN) 1
Pin(GPIO1, mode=IN) 0
Pin(GPIO1, mode=IN) 1
WORKING 283
Pin(GPIO1, mode=IN) 0
WORKING 284
WORKING 285
WORKING 286
```

robert-hh, 2023-03-29:
I changed the tcode a little bit, counting the irq's, and it works fine at 1 kHz.
```
from time import sleep_ms

pin = Pin(1, Pin.IN)
count = 0

def on_change(pin):
    global count
    state = disable_irq()
    pin.value()
    count += 1
    enable_irq(state)

pin.irq(on_change, Pin.IRQ_RISING | Pin.IRQ_FALLING)

for i in range(1000):
        idle()
        sleep_ms(100)
        print("WORKING", count)
```
The disable_irq()/eneable_irq() statements are NOT required, and I tested with and without. No change. The counter increases as expected,

GuillaumeLeclerc, 2023-03-30:
I know it's not required but I added it trying to solve the problem. To be honest I don't mind using the pin from the global context instead of the argument so it's not a big deal. If you think it's not an issue with Micropython and with my board/setup then feel free to close.

robert-hh, 2023-03-30:
I do not question your observation, but without being able to reproduce that behavior it is not possible to find the reason.