micropython/micropython #11148: docs: for time.localtime()

hasenradball, 2023-03-29:
Hi Together,

I would remark that this description in the Doc is a little bit missleading.

'''
time.mktime()
https://docs.micropython.org/en/latest/library/time.html
This is inverse function of localtime. It’s argument is a full 8-tuple which expresses a time as per localtime. It returns an integer which is the number of seconds since Jan 1, 2000.
''''

Because on the Pico W the RTC works with the epoch since 1970.
Maybe a small redesign of this description would be perfect, my opinion.

A other point is that I want to aks what the actual status is about:
time.gmtime()
time.localtime()

Is there actually a Timezone information available in micropython?
If not Iam right that these two functions leads to the same result in time structure?

Thanks
Frank

peterhinch, 2023-03-29:
There is no timezone support in MicroPython.

robert-hh, 2023-03-29:
@hasenradball, thank you for your comment. Since you see the need, please suggest a fix for the documentation by submitting a pull request for it.

jonnor, 2024-07-13:
Does anyone have concrete suggestions for improvements to the documentation? If not I suggest closing this issue, as the questions have been answered.

closed 2024-07-13 by peterhinch