micropython/micropython #11149: docs: WLAN.status- Decription of enum value

hasenradball, 2023-03-29:
Dear Support Team,

where can I find the implementation of the `wlan.status()` documented here:
[https://docs.micropython.org/en/latest/library/network.WLAN.html#network.WLAN.status](https://docs.micropython.org/en/latest/library/network.WLAN.html#network.WLAN.status)
```
WLAN.status([param])

    Return the current status of the wireless connection.

    When called with no argument the return value describes the network link status. The possible statuses are defined as constants:
            STAT_IDLE – no connection and no activity,
            STAT_CONNECTING – connecting in progress,
            STAT_WRONG_PASSWORD – failed due to incorrect password,
            STAT_NO_AP_FOUND – failed because no access point replied,
            STAT_CONNECT_FAIL – failed due to other problems,
            STAT_GOT_IP – connection successful.
 ```
 Because the assumption, that the list is ordered in ascending order of the status value seems not the case, right?
Or where can I look up the values of the enums?

The point is at my `Pi zero W`, I got `status == 3` if wlan is connected. It seems not to be the right value.

robert-hh, 2023-03-29:
The value of 3 matches the state `network.STAT_GOT_IP,` so it is the right value.

hasenradball, 2023-03-29:
@robert-hh 
Hi Robert how can I look up the int values of the constants, because it seems that the Docu list the enums not in ascending order, right?

would it not be the best notice also the int value of the network states in the docu?
So if it is so this chat or issue would not be necessary.
:-)

robert-hh, 2023-03-29:
```
import network
print(network.STAT_GOT_IP)

```
The int values may differ between ports. Therefore the symbolic values. I agree that there is no simple way to map them. You could create an dictionary for a mapping, if you like. eg.
```
status_names = {
    network.STAT_CONNECTING: 'STAT_CONNECTING',   
    network.STAT_CONNECT_FAIL: 'STAT_CONNECT_FAIL',  
    network.STAT_GOT_IP: 'STAT_GOT_IP',       
    network.STAT_IDLE: 'STAT_IDLE',         
    network.STAT_NO_AP_FOUND: 'STAT_NO_AP_FOUND',  
    network.STAT_WRONG_PASSWORD: 'STAT_WRONG_PASSWORD'
}
```
And I agree as well that this topic is more an discussion item than an issue.

jonnor, 2024-07-13:
There seems to be a resolution provided, so I am proposing to close this issue.

hasenradball, 2024-07-13:
@jonnor can you link the solition?

jonnor, 2024-07-13:
The solution to find the integer status values is to use the interpreter, as examplified by the code from @robert-hh above - https://github.com/micropython/micropython/issues/11149#issuecomment-1488196823

projectgus, 2025-02-04:
I think the initial issues' confusion was that the constants are defined in the `network` module.

The wording in this doc has been improved a little, it now says "When called with no argument the return value describes the network link status. The possible statuses are defined as constants in the [network](https://docs.micropython.org/en/latest/library/network.html#module-network) module:" instead of the previous "The possible statuses are defined as constants:"

closed 2025-02-04 by projectgus