micropython/micropython #11156: rp2: OPEN_DRAIN and PWM doesn't work

GuillaumeLeclerc, 2023-03-30:
```
from machine import Pin, PWM
PIN_MOTOR = 0

motor = Pin(PIN_MOTOR, Pin.OPEN_DRAIN, value=1)
pwm_motor = PWM(motor)
pwm_motor.freq(10_000)

pwm_motor.duty_u16(0)
pwm_motor.duty_u16(2**16)
```

According to my oscilloscope this set the pin to high instead of a drain. using `high()` and `low` without PWM works

robert-hh, 2023-03-30:
That's strange, since the PWM class configures the Pin itself. You do not have to set a specific mode.  Using just `Pin(PIN_MOTOR)` is sufficient. But I can check that.

GuillaumeLeclerc, 2023-03-30:
Well I **want** a specific mode. Without the PWM it works as expected:

- Low: Open_Drain
- High: High-Z

The problem is that with PWM when high I get 3.3V instead of High-Z and my motor doesn't turn off while it does when not using PWM.

robert-hh, 2023-03-30:
The PWM pins are always push/pull. To get the pin High-Z, you have to reconfigure the pin as machine.Pin object with the wanted properties. and configure it again as PWM when needed. Since at RP2 both Pin objects and PWM objects are singletons (pointers into a fixed table), that change will not create dynamic objects.

GuillaumeLeclerc, 2023-03-30:
So it's not possible to have PWM with High-Z/open_drain ?. I have to implement it in software myself with a timer?

robert-hh, 2023-03-30:
That's correct. The PWM is a hardware module and this behavior cannot be changed, as far as I read the data sheet. You can implement it in different ways. By hardware (and additional open collector driver consisting of a single transistor), by setting the pin to PIN mode when PWM is to be stopped. If you can accept the jitter, by timer.

I do not know at which voltage your motor runs. But even with a open drain setting of the output, the ESD diode would clamp the output level to about 4V.

closed 2023-03-30 by GuillaumeLeclerc