micropython/micropython #11166: Enable AP mode on ESP32 and enable TCP socket. After a few hours (overnight), the phone can connect to ESP32, but it cannot use TCP socket. It is preliminarily determined that the socket has been recycled by the system due to a long time of not connecting to ESP32 (gc has been disabled). The requirement of my current project is that the socket always exists and not be recycled by the system. How to set it up?

drc5521, 2023-03-31:
Enable AP mode on ESP32 and enable TCP socket. After a few hours (overnight), the phone can connect to ESP32, but it cannot use TCP socket. It is preliminarily determined that the socket has been recycled by the system due to a long time of not connecting to ESP32 (gc has been disabled). The requirement of my current project is that the socket always exists and not be recycled by the system. How to set it up?

jonnor, 2024-09-01:
gc has been disabled? You cannot run for many hours with a GC disabled, the system will run out of memory.

Furthermore, it cannot generally be guaranteed that a socket never goes down. There can be hardware or network failures, for example. Clients should handle disconnects and reconnects.

In order for us to do anything with an issue like this, a minimal code example that reproduces the issue needs to be provided.

projectgus, 2025-02-04:
Closing as we would need a code sample to debug this. @drc5521 if you're still trying to resolve this then suggest opening a new Discussion instead of an issue, and posting your code and as much detail as you can supply.

closed 2025-02-04 by projectgus