micropython/micropython #11181: docs:  full Documentation for the Pi Pico W

hasenradball, 2023-04-02:
a) When I want to use the following code on the Pico W:

''' 
>>> w = network.WLAN(network.STA_IF)
>>> w.status('rssi')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: unknown status param
''' 
I got an error!!!

But in the documentation is described exact this usage....

''' 
Return the current status of the wireless connection.
When called with no argument the return value describes the network link status. The possible statuses are defined as constants:
STAT_IDLE – no connection and no activity,
STAT_CONNECTING – connecting in progress,
STAT_WRONG_PASSWORD – failed due to incorrect password,
STAT_NO_AP_FOUND – failed because no access point replied,
STAT_CONNECT_FAIL – failed due to other problems,
STAT_GOT_IP – connection successful.
When called with one argument param should be a string naming the status parameter to retrieve. Supported parameters in WiFI STA mode are: 'rssi'.
'' 
b) Next issue:

when I use the wifi connect method with a wrong `password` in a network which I logged in recently with the right `password`, the login seems to be successfull!!!
WHY???
WTF happens here?
You have to give just the right 'ssid` to log in.
This is not the good behaviour I want to have!

Frank

PMG1230, 2024-02-26:
Hi Frank, 

I was stumped how the returned value of network.status() related to the constants as well.  It turns out you can get the translation via the REPL: 
>>>import network
>>>help(network)

that provides you the different status constants with values. 

Cheers,
Johannes

hasenradball, 2024-02-26:
@PMG1230 and about the wifi credentials?

jonnor, 2024-09-01:
Can you provide example code for reproducing the issue where using wrong password works?

hasenradball, 2024-09-02:
All I have is here:

when I use the wifi connect method with a wrong password in a network which I logged in recently with the right password, the login seems to be successfull!!!
WHY?
WTF happens here?

For example I use the following config for my connection:

config = {'ssid': 'default', 'key': 'default'}

If I set ONLY the right ssid and let key as default the pico connects!?!

You have to give just the right 'ssid` to log in.
This is not the good behaviour I want to have!

My example is here:
[example](https://github.com/hasenradball/PicoEx/blob/main/Pico_wifi_example.py)