micropython/micropython #11208: SSL Version Issue with wrap_socket

strunker, 2023-04-05:
So I am trying to create a simple https web server on an esp32, running latest stable version of Micropython 1.19.1.

The code runs on the device however if I try to use the python requests library on my laptop and issue a simple get to the esp web server I receive the below error. I can see the request hit the web server, and it responds, but there is something strange about the ssl/tls version that is being returned. On standard python, you can create an ssl context and specify min/max versions of tls but I dont see any way to achive this on micropython. 

Searched everywhere and it doesnt seem like most people care too much about using SSL on their projects but I need it for mine so this is causing me delay at this point. Hopefully I am just missing something basic here and this is a solvable problem. 

Thanks!

[image]

```
import machine
import ssl
import socket
try:
    addr = socket.getaddrinfo('0.0.0.0', 8443)[0][-1]
    s = socket.socket()
    s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
    s.bind(addr)
    s.listen(5)
except:
    pass

print('listening on', addr)

cert="cert.der"
key="key.der"
with open(cert, 'rb') as f:
    cert = f.read()
with open(key, 'rb') as f:
    key = f.read()

while True:
    cl, addr = s.accept()
    print('client connected from', addr)
    print("1")
    ssl.wrap_socket(cl, server_side=True,cert=cert,key=key)
    print("2")
    cl.send('HTTP/1.1 200 OK\n')
    cl.send('Content-Type: text/html\n')
    cl.send('Connection: close\n\n')
    cl.close()
```

dpgeorge, 2023-04-06:
There is an SSL server example at https://github.com/micropython/micropython/blob/master/examples/network/http_server_ssl.py which works, it runs on an esp32 (and the unix port, and other ports that support wifi/ethernet) and allows doing a request from normal Python.

Please try running that test.  If it doesn't work on v1.19.1 then please try the latest unstable firmware.

strunker, 2023-04-06:
I just wanted to say @dpgeorge  i caught one of your comments on this topic and it was helpful (there arent many I searched high and low), thanks for being amazing. This sample does work, and I wish I had found it sooner I likely would have saved myself a ton of stress. Examinng my code vs the sample I dont really understand why my code isnt working.

I refactored the sample to get rid of the byte string certs. And instead just read the cert content right in. If anyone else stumbles across this and wants to know how to generate the certs, the below is what I used. Since this is for internal use only right now I made the cert expiration rather long, 100 years. May not want to do that in prod. 

openssl req -x509 -newkey rsa:2098 -keyout key.der -out cert.der -sha256 -days 36500 -nodes

What I will say, doing a little more testing, if you call function main from below sample with False. Things do not work properly at all, so perhaps the code you have written within 'if use stream' conditional is key to making this work correctly. 

In any event this is great, how can we get this into the official micropython documentation the SSL webserver code is lacking/missing entirely. 

THANK YOU!

```
try:
    import usocket as socket
except:
    import socket
import ussl as ssl

cert="cert.der"
key="key.der"
with open(cert, 'rb') as f:
    cert = f.read()
with open(key, 'rb') as f:
    key = f.read()

CONTENT = b"""\
HTTP/1.0 200 OK

Hello #%d from MicroPython!
"""

def main(use_stream=True):
    s = socket.socket()

    # Binding to all interfaces - server will be accessible to other hosts!
    ai = socket.getaddrinfo("0.0.0.0", 8443)
    print("Bind address info:", ai)
    addr = ai[0][-1]

    s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
    s.bind(addr)
    s.listen(5)
    print("Listening, connect your browser to https://<this_host>:8443/")

    counter = 0
    while True:
        res = s.accept()
        client_s = res[0]
        client_addr = res[1]
        print("Client address:", client_addr)
        print("Client socket:", client_s)
        # CPython uses key keyfile/certfile arguments, but MicroPython uses key/cert
        client_s = ssl.wrap_socket(client_s, server_side=True, key=key, cert=cert)
        print(client_s)
        print("Request:")
        if use_stream:
            print("Using Stream")
            # Both CPython and MicroPython SSLSocket objects support read() and
            # write() methods.
            # Browsers are prone to terminate SSL connection abruptly if they
            # see unknown certificate, etc. We must continue in such case -
            # next request they issue will likely be more well-behaving and
            # will succeed.
            try:
                req = client_s.readline()
                print(req)
                while True:
                    h = client_s.readline()
                    if h == b"" or h == b"\r\n":
                        break
                    print(h)
                if req:
                    client_s.write(CONTENT % counter)
            except Exception as e:
                print("Exception serving request:", e)
        else:
            print("Not Using Stream")
            print(client_s.recv(4096))
            client_s.send(CONTENT % counter)
        client_s.close()
        counter += 1
        print()

main()
```

dpgeorge, 2023-04-06:
The problem with your code was this line:
```py
ssl.wrap_socket(cl, server_side=True,cert=cert,key=key)
```
It needs to be:
```py
cl = ssl.wrap_socket(cl, server_side=True,cert=cert,key=key)
```
Otherwise when you do `cl.send('HTTP/1.1 200 OK\n')` it's sending that data unencrypted over the original (unwrapped) socket and the other side is getting very confused.

> What I will say, doing a little more testing, if you call function main from below sample with False. Things do not work properly at all,

You are right, that part of the if-branch will only work on CPython.  We should probably get rid of that to avoid confusion.

> In any event this is great, how can we get this into the official micropython documentation?

We are working on that.  But you should always remember to check the examples, there are some good tips in that directory.

strunker, 2023-04-06:
> The problem with your code was this line:
> 
> ```python
> ssl.wrap_socket(cl, server_side=True,cert=cert,key=key)
> ```
> 
> It needs to be:
> 
> ```python
> cl = ssl.wrap_socket(cl, server_side=True,cert=cert,key=key)
> ```
> 
> Otherwise when you do `cl.send('HTTP/1.1 200 OK\n')` it's sending that data unencrypted over the original (unwrapped) socket and the other side is getting very confused.
> 
> > What I will say, doing a little more testing, if you call function main from below sample with False. Things do not work properly at all,
> 
> You are right, that part of the if-branch will only work on CPython. We should probably get rid of that to avoid confusion.
> 
> > In any event this is great, how can we get this into the official micropython documentation?
> 
> We are working on that. But you should always remember to check the examples, there are some good tips in that directory.

This is great, its pretty late here, but I will test my original code tomorrow and put together something that works in the way I require. I had a feeling it was something silly I was overlooking in syntax. Thanks again, and I will remember to check here. I have been reading the docs diligently but not really looking here for samples.

closed 2023-04-06 by strunker