micropython/micropython #11222: docs: How to use mem32 indexing for fast GPIO manipuation for all boards

GuillaumeLeclerc, 2023-04-08:
Hi!

From the documentation it's very well explained how to do this for the `stm` boards and the `viper` backend. Unfortunately I couldn't find how to do this for `rp2` and other boards.

Thanks for the awesome work!

rkompass, 2023-04-09:
Have a look at section 2.3.1 of the RP2040 datasheet: The SIO. For example to toggle  GPIO pin 7 in one cpu cycle you may write 1<<7 to SIO: GPIO_OUT_XOR which is done by `mem32[0xd000001c] = 1<<7`. (I didn't try this out though).

Could you move this to discussions? It's not an issue I would say.

GuillaumeLeclerc, 2023-04-09:
Isn't it a problem that it's documented for a single board and none of the others ?

the `stm` board has a bunch of constants defined to help with this but it's not available on any others

rkompass, 2023-04-13:
The `stm` module in the long run might be deprecated. Perhaps parts of it should prevail.
Seems like you are missing something like stm.GPIOA + stm.GPIO_ODR for the rp2.
Say rp2.SIO and rp2.GPIO_OUT, among others?
That should be specified and could make up an enhancement proposal.

GuillaumeLeclerc, 2023-04-14:
If the `stm` is going to be depreciated then I think it's ok to keep it this way, it was just confusing/frustrating to only see it for a single controller

closed 2023-04-14 by GuillaumeLeclerc