micropython/micropython #11237: RPi Pico behaves weird/frozen if using _thred library in uPython

bcseh202, 2023-04-11:
I was following the book called [Get Started with MicroPython on Raspberry Pi Pico][1] and i succesfully made it to the 65th page but i cannot continue.

I wrote the code just like in the book (i use VS Code and the Pico-W-Go extension not the Thonny IDE but i hadn't have any issues so far with it) but when i upload it or just simply run it on the Pico, the REPL freezes, the button for the pedestrian system doesn't work and i have to nuke.uf2 the pico and reflash before i can use it again. Everything other works fine, i read that multithreading is crap in uPython and my case is happened to others but i'm here for a solution. Anyone has even solved this or know other multithreading modes in uPython for Pico?

Also i got the Pylance error/warning ```"button_pressed" is assigned before global declaration Pylance``` for line 40 but it should be good code.

I knew C, but i want to play with other langauges too so i don't want to switch. :)

`
import machine
import utime
import _thread
#library for multithreading to work

led_red = machine.Pin(15, machine.Pin.OUT)
led_yellow = machine.Pin(14, machine.Pin.OUT)
led_green = machine.Pin(13, machine.Pin.OUT)
button = machine.Pin(28, machine.Pin.IN, machine.Pin.PULL_DOWN)
led_pedestrianAllow = machine.Pin(25, machine.Pin.OUT)

global button_pressed
#global variable for multithreading
button_pressed = False

def button_reader_thread():
    #defines the thread on the second core
    global button_pressed
    #allows the global variable to be written to and not only read it
    while True:
        if button.value() == 1:
            button_pressed == True
        utime.sleep(0.01)

_thread.start_new_thread(button_reader_thread, ())
#the thread need to be started manually

led_red.value(0)
led_yellow.value(0)
led_green.value(0)

while True:
    if button_pressed == True:
        led_red.value(1)
        for i in range(10):
            led_pedestrianAllow.value(1)
            utime.sleep(0.2)
            led_pedestrianAllow.value(0)
            utime.sleep(0.2)
        global button_pressed
        button_pressed = False
    led_red.value(1)
    utime.sleep(5)
    led_yellow.value(1)
    utime.sleep(5)
    led_red.value(0)
    led_yellow.value(0)
    led_green.value(1)
    utime.sleep(5)
    led_green.value(0)
    led_yellow.value(1)
    utime.sleep(5)
    led_yellow.value(0)
`
*Only thing i changed is the buzzer for the built in led because it was annoying.*

Also firmware file name: rp2-pico-20220618-v1.19.1.uf2
REPL version: MicroPython v1.19.1 on 2022-06-18; Raspberry Pi Pico with RP2040

  [1]: https://hackspace.raspberrypi.com/books/micropython-pico

peterhinch, 2023-04-12:
There seems to be [an issue](https://github.com/micropython/micropython/issues/10621#issuecomment-1501036389) with global variables and threads. Also this line does nothing
```python
global button_pressed
#global variable for multithreading
```
Python's `global` keyword is only meaningful within a function or method. I would replace that line with
```python
button_pressed = False  # Create the variable and assign a value
#global variable for multithreading
```
which will reserve globally accessible storage. In line with the reference I cited, you might like to declare a few more variables with global scope, access them from the thread, and see if thatmakes a difference.

Gadgetoid, 2023-04-19:
Possibly not the crux of your problem, but “button_pressed == True” in the “button_reader_thread” is an equality test, so your “button_pressed” value will never be set to true.

jonnor, 2024-09-02:
Is there a minimal reproducible example of this issue? It would make it much more possible to debug and maybe fix it.

closed 2025-02-04 by projectgus