micropython/micropython #11241: BLE.gap_connect() call results in "TypeError: function doesn't take keyword arguments" (ESP32-WROOM-32D (DFRobot DFR0575 board))

danielphili, 2023-04-11:
Hi everyone,

I am calling the following couple of lines in my micropython code:
`ble = bluetooth.BLE()
ble.gap_connect(addr_type, addr, max_conn_interval_us=5000)`
My intention is to increase throughput and according to the docs, this is a possible way to do that (at the cost of higher power consumption, which is acceptable for me).

Unfortunately, I receive an error message saying "TypeError: function doesn't take keyword arguments".
And indeed, when just calling `ble.gap_connect(addr_type, addr)` everything works just fine, but the reaction time is slow to my taste (around 100ms, so, improvable)..

My board is an ESP32-WROOM-32D (DFRobot DFR0575 board) running MicroPython V1.19.

I have read in other issues that some functions switched to positional arguments entirely, but entering more than three positional arguments (like `ble.gap_connect(addr_type, addr, 2000, 0, 5000)`, the 2000 in representing scan_duration_ms, 0 representing min_conn_interval_us, and 5000 my desired value for max_conn_interval_us, referring to the keywords mentioned in the online doc) results in `OSError: -17`

Here's a (not exactly minimal) MWE that replicates the issue connecting to the MicroPython kernel with JupyterNotebook:

```
%serialconnect to --port=COM14 --baud=115200
import bluetooth
import time
import binascii

_IRQ_SCAN_RESULT = const(5)
_IRQ_SCAN_DONE = const(6)
_IRQ_PERIPHERAL_CONNECT = const(7)

discovered_device_addr_list = list()
discovered_devices = dict()

current_conn_handle = 0
current_addr_type = 0
current_addr = 0

def raw_as_hex_bytes(raw_data):
    x = binascii.hexlify(raw_data).decode()
    return [x[2*(k-1):2*k] for k in range(1,len(x)//2 + 1)]

def irqhandler(event, data):
    global current_conn_handle
    global current_addr_type
    global current_addr
    if event == _IRQ_SCAN_RESULT:
        addr_type, addr, adv_type, rssi, adv_data = data
        addr_hex = ':'.join(raw_as_hex_bytes(addr))
        # add the scanned device to the devices dictionary
        if not addr_hex in discovered_device_addr_list:
            adv_data_hex = ''.join(raw_as_hex_bytes(adv_data))
            discovered_device_addr_list.append(addr_hex)
            discovered_devices[addr_hex] = {  
                'addr_type': addr_type,
                'addr': binascii.unhexlify(addr_hex.replace(':','')), #addr,
                'addr_hex': addr_hex,
                'adv_type': adv_type,
                'rssi': rssi,
                'adv_data': adv_data,
                'adv_data_hex': adv_data_hex,                    
                'adv_name': None,
                'device_nbr': len(discovered_device_addr_list),
                }
            print('Found device', discovered_devices[addr_hex]['device_nbr'], ':', addr_hex)

        elif event == _IRQ_SCAN_DONE:
            print('Scan ended.')
            
        elif event == _IRQ_PERIPHERAL_CONNECT:
            # Connect successful.
            current_conn_handle, current_ddr_type, current_addr = data
            print('Connected.')

ble = bluetooth.BLE()
ble.active(True)
ble.irq(irqhandler)

ble.gap_scan(2000, 30000, 30000)
time.sleep_ms(1000)
ble.gap_scan(None)

target_addr_hex = 'e8:7c:85:7e:a9:34'
if target_addr_hex in discovered_device_addr_list:
    target_device = discovered_devices[target_addr_hex]
    target_addr_type = target_device['addr_type']
    target_addr = target_device['addr']
    print('Try connecting to ', target_addr_hex, '...')
    
    
    ble.gap_connect(target_addr_type, target_addr, max_conn_interval_us=5000)
    
    
    time.sleep_ms(2000)
    ble.gap_disconnect(current_conn_handle)
```

Ryboster, 2023-08-10:
Hi, I've got one question. What method can I use to get the event? In the script you haven't passed any objects or values to irqhandler's "event" parameter.

I need to get a list of discovered devices returned via the use of BLE().gap_scan(). The documentation says you should use BLE().irq(), but it needs 2 arguments passed to it, one of them being the handler.

danielphili, 2023-08-11:
Hi,I register the previously defined irqhandler function with the BLE class instance as a callback function in case any irq event should occur (which is the case, for example, when a single scan result is obtained), by calling ble.irq(irqhandler). Now, the requirement on the function irqhandler is that it takes two positional arguments, namely, event and data, which will be filled by the BLE instance automatically. A function that returns you a list of discovered devices would not be a single function I think. It is almost like in my case, you need a centrally managed database (could be a global dict as in my case) that is extended with a scan result once it occurs. But, as I said, beware the number of devices can get very high and overload the uC, resulting in a crash.Hope this clears things up.

ekspla, 2024-08-22:
It seems that keyword arguments are not allowed in this method.  Please see modbluetooth.c in the source repository, as well as [an issue similar to yours.](https://github.com/micropython/micropython/issues/15418)

jonnor, 2024-09-02:
All the arguments to gap_connect() must be passed as positional arguments. In order, they are `addr_type, addr, scan_duration_ms, min_conn_interval_us, max_conn_interval_us`. To leave an argument as the default, pass `None`. This is from the source code of bluetooth_ble_gap_connect() in modbluetooth.c, as mentioned by @ekspla.

closed 2024-09-02 by jonnor