micropython/micropython #11247: [RP2-W] WLAN not usable, giving EPERM in every operation

ThomasChr, 2023-04-12:
So this seems to be a tough one for me.
I've got a Raspberry Pi Pico W. I've attached an BME280 Sensor via I2C, the connected Pins are:
```
BME VCC = Pico PinNo 36 (3V3(OUT))
BME GND = Pico PinNo 38 (GND)
BME SDA = Pico PinNo 11 (I2C0 SDA)
BME SCL = Pico PinNo 12 (I2C0 SCL)
```
And here is the code:
```
import time
import ustruct
import machine
import network
import usocket

# Globals
sensorid = 99
sensoraddr = 0x76
sendseconds = 60
secondstrywificonnect = 10
REG_DATA = 0xF7
REG_CONTROL = 0xF4
REG_CONFIG = 0xF5
REG_CONTROL_HUM = 0xF2
REG_HUM_MSB = 0xFD
REG_HUM_LSB = 0xFE
OVERSAMPLE_TEMP = 2
OVERSAMPLE_PRES = 2
MODE = 1

while True:
    try:
        # Connect to WiFi
        wlan = network.WLAN(network.STA_IF)
        if wlan.isconnected() == False:
            wlan.active(True)
            wlan.connect('FRITZ!Box 7490', 'x')
            second = 0
            while wlan.isconnected() == False:
                time.sleep(1)
                second += 1
                if second > secondstrywificonnect:
                    machine.reset()

        # Initialize I2C
        i2c = machine.I2C(0, sda=machine.Pin(8), scl=machine.Pin(9), freq=100000)

        # Set Oversampling
        OVERSAMPLE_HUM = 2
        i2c.writeto_mem(sensoraddr, REG_CONTROL_HUM, bytearray([OVERSAMPLE_HUM]))
        control = OVERSAMPLE_TEMP << 5 | OVERSAMPLE_PRES << 2 | MODE
        i2c.writeto_mem(sensoraddr, REG_CONTROL, bytearray([control]))

        # Read Calibration data from sensor
        cal1 = i2c.readfrom_mem(sensoraddr, 0x88, 24)
        cal2 = i2c.readfrom_mem(sensoraddr, 0xA1, 1)
        cal3 = i2c.readfrom_mem(sensoraddr, 0xE1, 7)

        # Convert bytes to words
        dig_T1 = ustruct.unpack('H', cal1[0:2])[0]
        dig_T2 = ustruct.unpack('h', cal1[2:4])[0]
        dig_T3 = ustruct.unpack('h', cal1[4:6])[0]
        dig_P1 = ustruct.unpack('H', cal1[6:8])[0]
        dig_P2 = ustruct.unpack('h', cal1[8:10])[0]
        dig_P3 = ustruct.unpack('h', cal1[10:12])[0]
        dig_P4 = ustruct.unpack('h', cal1[12:14])[0]
        dig_P5 = ustruct.unpack('h', cal1[14:16])[0]
        dig_P6 = ustruct.unpack('h', cal1[16:18])[0]
        dig_P7 = ustruct.unpack('h', cal1[18:20])[0]
        dig_P8 = ustruct.unpack('h', cal1[20:22])[0]
        dig_P9 = ustruct.unpack('h', cal1[22:24])[0]
        dig_H1 = ustruct.unpack('B', cal2[0:1])[0]
        dig_H2 = ustruct.unpack('h', cal3[0:2])[0]
        dig_H3 = ustruct.unpack('B', cal3[2:3])[0]
        dig_H4 = ustruct.unpack('b', cal3[3:4])[0]
        dig_H4 = (dig_H4 << 24) >> 20
        dig_H4 = dig_H4 | (ustruct.unpack('b', cal3[4:5])[0] & 0x0F)
        dig_H5 = ustruct.unpack('b', cal3[5:6])[0]
        dig_H5 = (dig_H5 << 24) >> 20
        dig_H5 = dig_H5 | (ustruct.unpack('B', cal3[4:5])[0] >> 4 & 0x0F)
        dig_H6 = ustruct.unpack('b', cal3[6:7])[0]

        # Wait in ms (Datasheet Appendix B: Measurement time and current calculation)
        wait_time = 1.25 + (2.3 * OVERSAMPLE_TEMP) + ((2.3 * OVERSAMPLE_PRES) + 0.575) + (
                (2.3 * OVERSAMPLE_HUM) + 0.575)
        time.sleep(wait_time / 1000)

        # Read temperature/pressure/humidity
        data = i2c.readfrom_mem(sensoraddr, REG_DATA, 8)
        pres_raw = (data[0] << 12) | (data[1] << 4) | (data[2] >> 4)
        temp_raw = (data[3] << 12) | (data[4] << 4) | (data[5] >> 4)
        hum_raw = (data[6] << 8) | data[7]

        # Refine temperature
        var1 = ((temp_raw >> 3 - dig_T1 << 1) * dig_T2) >> 11
        var2 = (((((temp_raw >> 4) - dig_T1) * ((temp_raw >> 4) - dig_T1)) >> 12) * dig_T3) >> 14
        t_fine = var1 + var2
        temperature = float(((t_fine * 5) + 128) >> 8)

        # Refine pressure and adjust for temperature
        var1 = t_fine / 2.0 - 64000.0
        var2 = var1 * var1 * dig_P6 / 32768.0
        var2 = var2 + var1 * dig_P5 * 2.0
        var2 = var2 / 4.0 + dig_P4 * 65536.0
        var1 = (dig_P3 * var1 * var1 / 524288.0 + dig_P2 * var1) / 524288.0
        var1 = (1.0 + var1 / 32768.0) * dig_P1
        if var1 == 0:
            pressure = 0
        else:
            pressure = 1048576.0 - pres_raw
            pressure = ((pressure - var2 / 4096.0) * 6250.0) / var1
            var1 = dig_P9 * pressure * pressure / 2147483648.0
            var2 = pressure * dig_P8 / 32768.0
            pressure = pressure + (var1 + var2 + dig_P7) / 16.0

        # Refine humidity
        humidity = t_fine - 76800.0
        humidity = (hum_raw - (dig_H4 * 64.0 + dig_H5 / 16384.0 * humidity)) * (
                dig_H2 / 65536.0 * (1.0 + dig_H6 / 67108864.0 * humidity * (1.0 + dig_H3 / 67108864.0 * humidity)))
        humidity = humidity * (1.0 - dig_H1 * humidity / 524288.0)
        if humidity > 100:
            humidity = 100
        elif humidity < 0:
            humidity = 0

        # Data is ready
        temperature = temperature / 100.0
        pressure = pressure / 100.0
        humidity = humidity

        # Send data to the Internet, a Post Request with http - we don't use SSL here!
        content = b'sensorid=' + str(sensorid) + '&temp=' + str(temperature) + '&press=' + str(
            pressure) + '&hum=' + str(humidity) + '&password=x'

        addr_info = usocket.getaddrinfo("x.de", 80)
        addr = addr_info[0][-1]
        sock = usocket.socket()
        sock.connect(addr)
        sock.send(b'POST /x.php HTTP/1.1\r\n')
        sock.send(b'Host: x.de\r\n')
        sock.send(b'Content-Type: application/x-www-form-urlencoded\r\n')
        sock.send(b'Content-Length: ' + str(len(content)) + '\r\n')
        sock.send(b'\r\n')
        sock.send(content)
        sock.send(b'\r\n\r\n')
        sock.close()

        # We're done
        time.sleep(sendseconds)
    except Exception as e:
        print(e)
        time.sleep(sendseconds)
```
(don't talk about code quality here - the code is old, and it is ugly as *** - I know...)
PS: This code runs on an ESP32 with Micropython without any problems whatsoever!

Firmware:
```
rp2-pico-w-20230411-unstable-v1.19.1-1014-gbde222ce8.uf2
```

And now the problem: When I send this code to the pico all runs fine, and is working like it should. UNTIL I remove power for even a short time. After that the WIFI is bricked and I can't do anything with it. Even resetting, rebooting, deinit() doesn't work anymore.
Here the status then:
```
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan
<CYW43 STA down 0.0.0.0>
>>> wlan.deinit()
>>> wlan.active(True)
>>> wlan
<CYW43 STA down 0.0.0.0>
>>> wlan.scan()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 1] EPERM

>>> wlan.active(False)
>>> wlan.deinit()
>>> wlan.active(True)
>>> wlan.scan()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 1] EPERM
>>> wlan.status()
0
```
(connect() does also fail, of course!)

There seems to be no way to get wifi running again. I think after some time (maybe the lease-time of the FritzBox?) it will work again - but there must be a way to get out of this!
I tried to reproduce it without the I2C Connected, but that doesn't give the same problems, however, after the problem has been triggered it doesn't mater if I2C is sill connected. It won't work anyway. Still I'm not sure what exactly triggers the problem, I can't test the whole code without I2C, so I can't rule that out reliably.

Even this simple code fails then:
```
import rp2
import time
import network
wid = "FRITZ!Box 7490"
wpass = "x"

secondstrywificonnect = 10

print(str(time.ticks_ms()) + ': Trying to connect')
# Connect to WiFi
wlan = network.WLAN(network.STA_IF)
if wlan.isconnected() == False:
 wlan.active(True)
 wlan.connect(wid, wpass)
 second = 0
 while wlan.isconnected() == False:
  time.sleep(1)
  second += 1
  print("Still trying")
  if second > secondstrywificonnect:
   print("Failed!")
else:
 print("Connected!")
 
#rp2.country('DE')
#wlan.deinit()
#wlan.config(pm = 0xa11140)
#wlan.active(True)
```
So I'm not quite sure how I got there, but it seems to be that there is a possibility to "brick" the Wifi.
Any ideas on how to approach that further?

Update: Deleting the Lease from the FritzBox does seem to help... but there must be a trick to do that from the Pico itself...

ThomasChr, 2023-04-12:
I can't recreate the problem with a Ruckus WIFI, so it has to be something with the FritzBox and not removing the lease!

ThomasChr, 2023-04-13:
The same happens also when I didn't connect to the wifi (wrong password). Still, after a reboot the Wifi only says "EPERM" and is absolutely unusable. I'm a little bit puzzled why I'm the only one experiencing this, but to me it seems that this is a major issue. Any help would be greatly appreciated!

ThomasChr, 2023-04-13:
Forumsthread: https://forums.raspberrypi.com/viewtopic.php?p=2098898

ThomasChr, 2023-04-23:
I now can debug it, and the problem seems to be that I can't activate the WiFi-Chip, here is the stack trace with an error:
[code]
_exit runtime.c:186
__assert_func runtime.c:270
cyw43_spi_init cyw43_bus_pio_spi.c:104
cyw43_ll_bus_init cyw43_ll.c:1471
cyw43_ensure_up cyw43_ctrl.c:187
cyw43_wifi_on cyw43_ctrl.c:469
cyw43_wifi_set_up cyw43_ctrl.c:549
network_cyw43_active network_cyw43.c:125
fun_builtin_var_call objfun.c:123
mp_call_function_n_kw runtime.c:695
mp_call_method_n_kw runtime.c:711
mp_execute_bytecode vm.c:1042
fun_bc_call objfun.c:278
mp_call_function_n_kw runtime.c:695
mp_call_function_0 runtime.c:669
parse_compile_execute pyexec.c:123
pyexec_friendly_repl pyexec.c:675
main main.c:194
platform_entry crt0.S:261
[/code]

ThomasChr, 2023-04-23:
Removing the assert leads to the following panic:
```
_exit runtime.c:186
panic runtime.c:325
_hw_endpoint_buffer_control_update32 rp2040_usb.c:91
_hw_endpoint_buffer_control_set_value32 rp2040_usb.h:90
_hw_endpoint_start_next_buffer rp2040_usb.c:183
hw_endpoint_xfer_start rp2040_usb.c:205
hw_endpoint_xfer dcd_rp2040.c:188
dcd_set_address dcd_rp2040.c:416
process_control_request usbd.c:667
tud_task_ext usbd.c:521
usbd_task mp_usbd.c:40
cyw43_delay_ms cyw43_configport.h:116
cyw43_ll_wifi_on cyw43_ll.c:1936
cyw43_wifi_on cyw43_ctrl.c:480
cyw43_wifi_set_up cyw43_ctrl.c:549
network_cyw43_active network_cyw43.c:125
fun_builtin_var_call objfun.c:123
mp_call_function_n_kw runtime.c:695
mp_call_method_n_kw runtime.c:711
mp_execute_bytecode vm.c:1042
fun_bc_call objfun.c:278
mp_call_function_n_kw runtime.c:695
mp_call_function_0 runtime.c:669
parse_compile_execute pyexec.c:123
pyexec_file pyexec.c:686
pyexec_file_if_exists pyexec.c:698
main main.c:182
platform_entry crt0.S:261
```

dpgeorge, 2023-04-24:
> I now can debug it, and the problem seems to be that I can't activate the WiFi-Chip, here is the stack trace with an error:

From your trace and debug output, it looks like `cyw43_spi_init()` is being called with `self->bus_data` not NULL.  But this should be impossible, because `cyw43_spi_init()` is only called if `cyw43_poll == NULL`, and that only happens when the cyw43 driver is fully reset, eg at power on or after calling `cyw43_deinit()`.

If you do a power cycle (completely remove power from the Pico W and wait long enough, eg 60 seconds) then all state should be reset in the Pico W and it should be able to join the WiFi again.  Unless maybe the FritzBox is causing some internal buffer overflow in the cyw43 driver upon rejoining??

It's also strange that the issue is only seen when you use I2C.  Is the issue there if you don't use I2C but do still send data over a socket?  It may be that the act of sending data triggers something in the FritzBox that causes the issue.

Can you also try the following:
- connect as usual to the FritzBox to trigger the problem
- try to reconnect tothe FritzBox and get the EPERM error
- fully power cycle the Pico W
- try to connect to another WiFi access point (not the FritzBox)

If that works then there must be some state in the FritzBox that's causing the Pico W to fail when starting up.

ThomasChr, 2023-04-24:
I tried to trigger it without the I2C Bus involved, and wasn't able to.
BUT it did hang in the join-State of the Wifi forever.

I removed the power for 120 seconds and the error finally vanishes completely.

I also managed to get the following crash sometimes:
```
_exit runtime.c:186
panic runtime.c:325
_hw_endpoint_buffer_control_update32 rp2040_usb.c:91
_hw_endpoint_buffer_control_set_value32 rp2040_usb.h:90
_hw_endpoint_start_next_buffer rp2040_usb.c:183
hw_endpoint_xfer_start rp2040_usb.c:205
hw_endpoint_xfer dcd_rp2040.c:188
dcd_set_address dcd_rp2040.c:416
process_control_request usbd.c:667
tud_task_ext usbd.c:521
usbd_task mp_usbd.c:40
cyw43_delay_ms cyw43_configport.h:116
cyw43_ll_bus_init cyw43_ll.c:1731
cyw43_ensure_up cyw43_ctrl.c:187
cyw43_wifi_on cyw43_ctrl.c:469
cyw43_wifi_set_up cyw43_ctrl.c:549
network_cyw43_active network_cyw43.c:125
fun_builtin_var_call objfun.c:123
mp_call_function_n_kw runtime.c:695
mp_call_method_n_kw runtime.c:711
mp_execute_bytecode vm.c:1042
fun_bc_call objfun.c:278
mp_call_function_n_kw runtime.c:695
mp_call_function_0 runtime.c:669
parse_compile_execute pyexec.c:123
pyexec_file pyexec.c:686
pyexec_file_if_exists pyexec.c:698
main main.c:182
platform_entry crt0.S:261
```
In Micropython without the debugger you can only see that it hangs endlessly in trying to connect to the WiFi.
This seems to be because the debugger connection does make a clean reset, which resets the bus-state. The normal Micropython Reset doesn't seem to reset the bus state!
But it seems that reseting only the bus state does not be everything needed to remove the error.

So in my opinion the code is still cursed, and it seems that it has something to do with the I2C AND the FritzBox.
So for the next try I tried to connect to some other WiFi afterwards - my iPhone reports the device as connected, BUT Micropython stays in wifi.status() '1' - which means it doesn't move on.
```
>>> import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.connect('iPhone', 'abcdefgh')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 1] EPERM
>>> wlan.active(True)
>>> wlan.connect('iPhone', 'abcdefgh')
>>> wlan.status()
1
>>> wlan.isconnected()
False
>>> import time
>>> time.sleep(60)
>>> wlan.isconnected()
False
```
The iPhone does report the Pico as connected all the time, whatever that really means.

It seems to be a really tough problem and I really want to help more. I've got all the debugging stuff ready but it's hard to know where to start.
Also the "divide and conquer" approach isn't easy with some state beeing reset and some not. So I also can't be sure if all my above tests are absolutely correct!

ThomasChr, 2023-04-24:
If it helps you can connect to my computer and try it out for yourself.

dpgeorge, 2023-04-24:
> I removed the power for 120 seconds and the error finally vanishes completely.

So this indicates that either:
1. There is state in the RP2040 or CYW43439 (the WiFi chip) which needs to be reset and requires a proper power cycle to reset it.
2. After 120 seconds the FritzBox clears any state it has about the Pico W.  If it sees WiFi activity from it before then (but after the Pico W reset) then the FritzBox assumes that it's the same WiFi session with the Pico W and keeps some state around.

I would think it's option (1), but it's still hard to understand why because all bss/data state is reset on the RP2040, and the CYW43439 is also reset by its reset pin when the Pico W restarts.

> So for the next try I tried to connect to some other WiFi afterwards - my iPhone reports the device as connected, BUT Micropython stays in wifi.status() '1' - which means it doesn't move on

That means the Pico W is still trying to associate with the access point (the phone in this case) and hasn't yet asked for an IP address.  This would point to there being something wrong with the state of the Pico W, rather than the FritzBox.

One thing you can try (if you can build firmware yourself) is to increase the time that the CYW43439 is reset.  In `lib/cyw43-driver/src/cyw43_ctrl.c change as follows:
```diff
diff --git a/src/cyw43_ctrl.c b/src/cyw43_ctrl.c
index 808e1b2..021d612 100644
--- a/src/cyw43_ctrl.c
+++ b/src/cyw43_ctrl.c
@@ -167,9 +167,9 @@ static int cyw43_ensure_up(cyw43_t *self) {
 
     // Reset and power up the WL chip
     cyw43_hal_pin_low(CYW43_PIN_WL_REG_ON);
-    cyw43_delay_ms(20);
+    cyw43_delay_ms(200);
     cyw43_hal_pin_high(CYW43_PIN_WL_REG_ON);
-    cyw43_delay_ms(50);
+    cyw43_delay_ms(500);
 
     #if !CYW43_USE_SPI
     // Initialise SDIO bus
```

ThomasChr, 2023-04-24:
Bisecting:
```
20/50			-> fail
20/250			-> fail
100/250			-> fail
125/250			-> works
125/312			-> works
150/375			-> works
200/500			-> works
```
So it seems that 
```
    // Reset and power up the WL chip
    cyw43_hal_pin_low(CYW43_PIN_WL_REG_ON);
    cyw43_delay_ms(125);
    cyw43_hal_pin_high(CYW43_PIN_WL_REG_ON);
    cyw43_delay_ms(250);
```
does the trick.

And the other problem with the endless connection, I get that output, does that help?
```
83828: Trying to connect
[   83853] ETHTX len=350 dst=ff:ff:ff:ff:ff:ff src=28:cd:c1:09:2a:8b type=IPv4 srcip=0.0.0.0 dstip=255.255.255.255 UDP srcport=68 dstport=67 DHCPC 01 01 06 00 ed c4 9b e2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 28 cd c1 09 2a 8b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 opts: DISCOVER data= 63 82 53 63 35 01 01 39 02 05 dc 37 04 01 03 1c 06 ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40 1f 00 00 00 00 00 00
```
The problem is, that most of the time I do not get any output, that's the test code I'm using:
```
import network
wlan = network.WLAN(network.STA_IF)
wlan.deinit()
wlan.config(trace=15)
wlan.active(True)
wlan.connect('FRITZ!Box 7490', 'x')
```

dpgeorge, 2023-04-25:
With the timeout, can you please try smaller values for the second one, eg 125/50?

The ETHTX you see is the Pico W trying to send out a DHCP client request, to obtain an IP.  If you don't see any ETHRX then there's nothing coming back over the WiFi.  And if you don't see any ASYNC lines then that means the WiFi driver itself is not processing any events.

ThomasChr, 2023-04-25:
I really tried, but the lowest I got was 125/250.
I feel not good about the fact that everyone has to wait nearly half a second on every boot because FritzBox is crap. Maybe we can move this long timeout in the deinit() of a WiFi Object? Then it's only triggered if you explicitly call it. I would be fine with that.

maxi07, 2023-07-24:
@ThomasChr Did you ever resolve this issue? I am having similar issues, also working with a FritzBox.

ThomasChr, 2023-07-25:
@maxi07 I kind of resolved it with some ridiculously high timeouts (we're talking about near to half a second) when initializing the WiFi Stack. I'm not surprised that @dpgeorge doesn't want that high timeouts for everyone because of the Fritzbox. At the moment I'm still using my ESP32 which don't have the problem. A pity :-(

hoihu, 2023-11-06:
> I'm a little bit puzzled why I'm the only one experiencing this, but to me it seems that this is a major issue. Any help would be greatly appreciated!

I'm facing a similar issue, using the minimal example (using the nightly build as of today). I have a larger Application that I download (using 2 PIO SM's and one I2C). When exiting the app via CTRL+C and trying the following:

```
sta = network.WLAN(network.STA_IF)
sta.active(True)
sta.scan()
```

throws
```OSError: [Errno 1] EPERM```

I also see sometimes that `sta.scan()` completly freezes the system (can't CTRL+C it). I have no fritzbox but a Cisco AP.

The above script works on a vanilla PicoW Modul with no scripts installed btw.

Unfortunately so far, I can't pin it to a specific modul that I'm using...  The application that I use works on a normal Pico (without Wifi). I also made sure that I'm not using pins 23,24,25 and 29 (wlan pins).

hoihu, 2023-11-06:
ah. I see that the cyw43 driver uses PIO 1! So that makes a big difference and may be the reason why it was failing...

hoihu, 2023-11-06:
ok. That was the issue in my case: I used PIO1 and also DMA channels that are conflicting with the cyw43 wlan driver. So it's probably not related to the issue here (sorry for hijacking it..).

Maybe a small note in online doc might be helpful: "Warning: Wlan driver uses PIO1 and therefore PIO1 should not be reconfigured"

Also the DMA channel assignement might be helpful to note, although this is dynamically done via the SDK's `hw_claim_unused_from_range` function. But still it would be helpful to know that the wlan driver uses a low dma channel number, so that other, manual DMA channel selection should be done from top down (e.g. from 11 -> 0). Even better would be if that claim function would be available via python, so that an application that uses DMA could claim a number.