micropython/micropython #11251: esp32: Add initial support for ESP8685(base ESP32C3) SoCs

Rookior, 2023-04-13:


mcauser, 2023-04-13:
https://esp32.com/viewtopic.php?t=25384

> ESP8685 has an ESP32-C3 silicon die packaged together with 2MB flash in a 4x4 mm QFN-28 package. This package doesn't have SPI Flash signals bonded out to package pads.
For comparison, ESP32-C3F2 also has an ESP32-C3 die and 2MB flash, but the package is 5x5mm QFN-32, and SPI Flash signals are bonded out.
For software developers, ESP32-C3F2 and ESP8685 are equivalent, you can use "esp32c3" target in ESP-IDF to compile software for them.

> ESP8684 has an ESP32-C2 silicon die packaged together with (some amount of) flash in a 4x4 mm QFN-28 package. Just like the ESP8685, this package doesn't have SPI flash signals bonded out. ESP32-C2 is a lower-end chip compared to ESP32-C3, with less RAM and peripherals. Support for this chip is being added to ESP-IDF at the moment. I can't say yet whether ESP32-C2 will be marketed and sold only as part of ESP8684 package or separately as well.

ESP8685 is like a ESP32-C3 with no ext SPI flash.

ESP8684 is like a ESP32-C2 with no ext SPI flash.

Rookior, 2023-04-13:
**esp8685_datasheet_cn**

https://www.espressif.com.cn/sites/default/files/documentation/esp8685_datasheet_en.pdf

**Some modules datasheet using this chipesp8685**

https://www.espressif.com.cn/sites/default/files/documentation/esp8685-wroom-06_datasheet_en.pdf

Rookior, 2023-04-13:
发现是我使用的 esptool 版本的问题,采用最新版本后,既可以使用esp32-c3 固件,并不需要重新适配.自行关闭该issue.

Rookior, 2023-04-13:
I found that there is an issue with the esptool version I am using. With the latest version, I can use the esp32-c3 firmware without the need to re adapt. Close this issue

closed 2023-04-13 by Rookior