micropython/micropython #11253: On RP2 sys.stdin.read(64) and sys.stdin.readline() do not work if we have multiple of 64 bytes in buffer

pchala, 2023-04-13:
MicroPython v1.19.1-994-ga4672149b on 2023-03-29; Raspberry Pi Pico with RP2040

Pico code:
```
import sys

while True:
    line = sys.stdin.read(64)
    print(len(line))

```

pchala, 2023-04-13:
Even this code will stuck

```
import sys

while True:
    line = sys.stdin.read(2)
    print(len(line))
```

dpgeorge, 2023-04-14:
The above examples both work for me, with version `MicroPython v1.19.1-1015-g0acc73344 on 2023-04-14; Raspberry Pi Pico with RP2040`.  I ran the code and held down the space key and the output would print 64, 64, 64, ... or 2, 2, 2, ...

Are you using USB as the serial port?

pchala, 2023-04-14:
Indeed the way you deliver data is quite essential.

Here is my Windows PC code:

```
from serial import Serial

serial = Serial('COM5', baudrate=4_000_000)

serial.write(b'\x46'*64) 
print(serial.readline())

```

robert-hh, 2023-04-14:
Does it happen at a lower baud rate as well? I recall having tested the initial RP2 IRQ based UART implementation at 4MBit quite a bit, and it worked well. Later, the simple locking mechanisms was replaced by semaphores. The simple locking was not thread safe. Then the maximal baud rate it could handle decreased, but I cannot recall to which number.

olivierdekkers, 2023-04-14:
We have seen it happen on the default settings as well, that's how we first found it

pchala, 2023-04-14:
Communication in my case done through USB CDC. I think baud rate is ignored in this case.

pchala, 2023-04-14:
It may be in tinyusb or in REPL -> sys.stdin
When I start Thonny after failed test I see that 64 chars was in the buffer
[image]

robert-hh, 2023-04-14:
Forget my comment. I was talking about UART, but it's a tinyusb topic.

dpgeorge, 2023-04-14:
I can reproduce this issue, by using the above script on the host PC to send exactly 64 bytes at a time.  The Pico does not see any data until 256 bytes have been sent, then it sees all 256 bytes at once.

Actually the Pico does get the 64 byte chunks but TinyUSB does not get them from the low-level USB routines until 256 total bytes have been received.  It's related to the ZLP scenario (zero length packet) where if exactly the endpoint size (64 bytes in this case) worth of data is sent then the USB driver waits for more data.

As far as I can tell the bug is due to #7492, commit 0613d3e3561a82fffa36594647ef916b19bc912b.  That configures the endpoint size to be 256 bytes and TinyUSB does not see any data until this buffer is full (because TinyUSB requests that it be filled to capacity).  Except, in the case that a short packet is received from the host (ie packet length < 64 bytes) then the data is passed up to TinyUSB because that case is treated as "host has finished sending".

The fix to the current issue seems to be to revert the above commit, ie make CFG_TUD_CDC_EP_BUFSIZE match the endpoint size of 64 bytes. I'm not sure why @hathach suggested to increase this beyond the endpoint size, see comment https://github.com/micropython/micropython/issues/7479#issuecomment-873623632

@robert-hh reverting that commit will perhaps slow down the USB transfer (host -> device, ie OUT) a bit, but I think that's unavoidable because there's definitely a bug here.

robert-hh, 2023-04-14:
> everting that commit will perhaps slow down the USB transfer (host -> device, ie OUT) a bit,

I do not expect that to be a problem. As far as I know, no assumption is made about the USB transfer rate.

dpgeorge, 2023-04-21:
Fixed by ec1eeccab45af6893f19cd07292808bbe7c62cf5

closed 2023-04-21 by dpgeorge

hathach, 2023-04-25:
@dpgeorge sorry for late response, I will add a bit of explanation here. For a stream like CDC, when an muliple bytes of endpoints size (64 bytes) is transfer e.g 256 byte, TinyUSB currently expect an zero-length data to end otherwise it will assume host is sending more data. Normally these are taken care by termianl software. For example, there is no point to end transfer with each 64 bytes with MassStorage since they often queued with 512 bytes at minimum. Granted that MSC queues exact bytes thank to SCSI Command Block, though low level DCD doesn't have knowlege of class driver. 

Admittedly, there is defintely something we could improve here, I made an issue https://github.com/hathach/tinyusb/issues/2041 for this scenario and will push update when it is "fixed". Note: for rp2040 specifically, `CFG_TUD_CDC_EP_BUFSIZE` may get irrelavant in the future when I plan to queue bytes from fifo directly (will post update as well).