micropython/micropython #11257: Esp32-C3 i2c timeout error

macraj, 2023-04-14:
### Discussed in https://github.com/orgs/micropython/discussions/11252

<div type='discussions-op-text'>

<sup>Originally posted by **macraj** April 13, 2023</sup>
Hi, I have ESP-C3-01M-Kit. After deepsleep restart, I've get such error : E (224934) i2c: i2c_set_timeout(817): i2c timing value error, however i2c seems works properly. 
When I power reset module, no error but after one loop it's showing again. 
I tired changing i2c pins to 0 and 1 or  3 and 4 but all the same.
```py
from machine import Pin, I2C, deepsleep
from time import sleep
import bme280

I2C_LL_MAX_TIMEOUT = 10
i2c = I2C(0, freq=100_000)

bme = bme280.BME280(i2c=i2c)
sleep(0.1)
print(i2c.scan())

def main():
    
    print(bme.temperature)
    print(bme.humidity)
    print(bme.pressure)
    sleep(.1)
    i2c = None
    deepsleep(10000)
    
if __name__ == "__main__":
    main()
```

> Here output from Thonny:
```
MPY: soft reboot
[118]
22.24
33.48
980.50
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x5 (DSLEEP),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0xe3c
load:0x403ce000,len:0x6f4
load:0x403d0000,len:0x28d8
entry 0x403ce000
**E (586) i2c: i2c_set_timeout(817): i2c timing value error**
[118]
22.29
31.98
980.47
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x5 (DSLEEP),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0xe3c
load:0x403ce000,len:0x6f4
load:0x403d0000,len:0x28d8
entry 0x403ce000
**E (586) i2c: i2c_set_timeout(817): i2c timing value error**
[118]
22.34
31.61
980.42
.....
```</div>

mgsb, 2023-04-14:
What verson/commit are you using? There was a fix for ESP32-C3 I2C last year 12f99481518b0ebcb14f00b2323865a845c2a4f1

jonnor, 2024-09-01:
Most likely fixed by the above.

closed 2024-09-01 by jonnor