micropython/micropython #11269: docs: Building embedding example using shared/static lib with cmake (with example)

shakfu, 2023-04-15:
I was recently interested to build static and shared library versions of the `micropython/examples/embedding` case using cmake and I managed to get it roughly done ( see this [public repo](https://github.com/shakfu/micropython-embed-cmake) in case anyone is interested).

Along the way I found that micropython has quite a number of cmake-related files in its repo (see below). 

Since I did not use any of these in my above example, is there any documentation guidance on building MicroPython with cmake that I can use to improve my example? Thanks!

```text
ports/esp32/CMakeLists.txt
ports/esp32/main/CMakeLists.txt
ports/rp2/CMakeLists.txt
ports/zephyr/CMakeLists.txt

AND

examples/usercmodule/cexample/micropython.cmake
examples/usercmodule/cppexample/micropython.cmake
examples/usercmodule/micropython.cmake
extmod/extmod.cmake
extmod/nimble/nimble.cmake
ports/esp32/boards/ESP32_S2_WROVER/mpconfigboard.cmake
ports/esp32/boards/GENERIC/mpconfigboard.cmake
ports/esp32/boards/GENERIC_C3/mpconfigboard.cmake
ports/esp32/boards/GENERIC_C3_USB/mpconfigboard.cmake
ports/esp32/boards/GENERIC_D2WD/mpconfigboard.cmake
ports/esp32/boards/GENERIC_OTA/mpconfigboard.cmake
ports/esp32/boards/GENERIC_S2/mpconfigboard.cmake
ports/esp32/boards/GENERIC_S3/mpconfigboard.cmake
ports/esp32/boards/GENERIC_S3_SPIRAM/mpconfigboard.cmake
ports/esp32/boards/GENERIC_S3_SPIRAM_OCT/mpconfigboard.cmake
ports/esp32/boards/GENERIC_SPIRAM/mpconfigboard.cmake
ports/esp32/boards/GENERIC_UNICORE/mpconfigboard.cmake
ports/esp32/boards/LILYGO_TTGO_LORA32/mpconfigboard.cmake
ports/esp32/boards/LOLIN_C3_MINI/mpconfigboard.cmake
ports/esp32/boards/LOLIN_S2_MINI/mpconfigboard.cmake
ports/esp32/boards/LOLIN_S2_PICO/mpconfigboard.cmake
ports/esp32/boards/M5STACK_ATOM/mpconfigboard.cmake
ports/esp32/boards/OLIMEX_ESP32_POE/mpconfigboard.cmake
ports/esp32/boards/SIL_WESP32/mpconfigboard.cmake
ports/esp32/boards/UM_FEATHERS2/mpconfigboard.cmake
ports/esp32/boards/UM_FEATHERS2NEO/mpconfigboard.cmake
ports/esp32/boards/UM_FEATHERS3/mpconfigboard.cmake
ports/esp32/boards/UM_PROS3/mpconfigboard.cmake
ports/esp32/boards/UM_TINYPICO/mpconfigboard.cmake
ports/esp32/boards/UM_TINYS2/mpconfigboard.cmake
ports/esp32/boards/UM_TINYS3/mpconfigboard.cmake
ports/rp2/boards/ADAFRUIT_FEATHER_RP2040/mpconfigboard.cmake
ports/rp2/boards/ADAFRUIT_ITSYBITSY_RP2040/mpconfigboard.cmake
ports/rp2/boards/ADAFRUIT_QTPY_RP2040/mpconfigboard.cmake
ports/rp2/boards/ARDUINO_NANO_RP2040_CONNECT/mpconfigboard.cmake
ports/rp2/boards/GARATRONIC_PYBSTICK26_RP2040/mpconfigboard.cmake
ports/rp2/boards/NULLBITS_BIT_C_PRO/mpconfigboard.cmake
ports/rp2/boards/PICO/mpconfigboard.cmake
ports/rp2/boards/PICO_W/mpconfigboard.cmake
ports/rp2/boards/PIMORONI_PICOLIPO_16MB/mpconfigboard.cmake
ports/rp2/boards/PIMORONI_PICOLIPO_4MB/mpconfigboard.cmake
ports/rp2/boards/PIMORONI_TINY2040/mpconfigboard.cmake
ports/rp2/boards/SPARKFUN_PROMICRO/mpconfigboard.cmake
ports/rp2/boards/SPARKFUN_THINGPLUS/mpconfigboard.cmake
ports/rp2/boards/W5100S_EVB_PICO/mpconfigboard.cmake
ports/rp2/boards/W5500_EVB_PICO/mpconfigboard.cmake
ports/rp2/boards/WEACTSTUDIO/mpconfigboard.cmake
py/mkrules.cmake
py/py.cmake
py/usermod.cmake
```

brianredbeard, 2023-05-26:
Caveat: Like yourself I am more a consumer of the project as opposed to a contributor.  That said, since it's been ~6 weeks I'll take a stab at answering and more proficient users can correct me in the future.

There are a number of microcontroller projects who's build systems are complex CMake based mechanisms.  Your example cites two prominent platforms: ESP32 & the RP2 (rp2040).  The build systems for these platforms have slowly been incorporated into the project.  An example is the work tracked in #6892  (https://github.com/micropython/micropython/commit/66098c09850ccc31b49b341e7eb7a5f8526e359d).  

At the present time there is not explicit documentation on the consumption of CMake recipes in downstream projects which has then lead to some required messiness in the past, e.g.  [badgePython@badgeteam:/components/micropython](https://github.com/badgeteam/badgePython/tree/5a549a6ff77253821a3112300e248e2a76ebc158/components/micropython).  I know this first hand because *I am pursuing the same goal*, spurring from challenges with some of the provided APIs ([Discussion: Expand ESP32 NVS to support partition by label](https://github.com/orgs/micropython/discussions/9928)).

shakfu, 2023-05-27:
Thanks for your background on this. I hope the documentation improves at some point, especially for those project consumers who tend to use cmake in their downstream projects.

jonnor, 2024-08-24:
The documentation is in the git repository, in the README.md files, as well as the doc/ directory (Sphinx project).
Contributions are very welcomed!