micropython/micropython #11275: PICO pio exec jmp

webdeb, 2023-04-16:
Describe the bug
When you try to execute sm.exec("jmp(\"mylabel\")") from a micropython program the runtime throws an error:

Tried with different strings

```
# this throws IOASMError: expecting exactly 1 instruction
sm.exec("jmp(\"on\")")

# this throws NameError: name 'on' isn't defined
sm.exec("jmp(on)")

# and this (label is on instruction 30) throws
# PIOASMError: unknown label 30
sm.exec("jmp(30)")
```

Program to reproduce (initially found the bug in wokwiki, but the same happens on hardware)
https://wokwi.com/projects/362125235929964545

Expected behavior
It should be possible to execute a jmp instruction

Gadgetoid, 2023-04-19:
Looks like emitting a single “jmp” instruction without the context of existing labels is currently impossible. See: https://github.com/micropython/micropython/blob/b525f1c9ec8ffa9009754578932f3fad5f63026b/ports/rp2/modules/rp2.py#L288

I guess it compiles to 0 instructions, throwing the misleading “expecting exactly 1 instruction” error.

webdeb, 2023-04-19:
> Looks like emitting a single “jmp” instruction without the context of existing labels is currently impossible. See: [...]

Thank you for clarification. 👍

webdeb, 2023-04-23:
OK, I would like to describe my case:

I am loading some data into the fifo, and instead of performing multiple instructions from outside of the program, I let the pio program setup the registers in a "load" block. I found a workaround with a conditional jmp and nilling some of the params inside the program, but it would be more nice and clean to just do this:

```
put(data)
exec("jmp(load)")
```

webdeb, 2023-04-23:
An idea, _I dont know if its possible or not_, would be to have the pointers to the instructions mapped on the statemachine instances, so we could reuse this context for further instructions

```
sm = StateMachine(...)
sm.exec(f"jmp({sm.labels['load']})")

# OR just

sm.jmp("load")
```

Gadgetoid, 2023-04-24:
It's certainly *possible* but looks like it requires a non-trivial refactor of `StateMachine` to have any awareness of how the code is compiled, and any ongoing state at all. Right now `StateMachine` just gets what's effectively a precompiled binary, though that "binary` does include some  "metadata". Which makes me wonder.

What if we (and by "we" I am not implying that I am a MicroPython developer with any sway over this code, but rather referring to you and I) could emit the labels dict as part of the PIO program.

For the `blink_1hz` program (from `pio_1hz.py` in MicroPython's examples) that would change the output of `asm_pio` from:

```python
[array('H', [49168, 57345, 58687, 48450, 67, 41026, 57344, 58687, 48450, 72]), -1, -1, 36864, 0, None, 2, None]
```

to:

```python
[array('H', [49168, 57345, 58687, 48450, 67, 41026, 57344, 58687, 48450, 72]), -1, -1, 36864, 0, None, 2, None, {'delay_low': 8, 'delay_high': 3}]
```

This adds a `dict` where the key is the label and the value is the instruction number within that program assuming it's stored at offset 0 in the SM instruction memory. Internally it just moves the already used dict from being a class-variable to a member of the `prog` list so it's implicitly returned as part of the "compiled program"- https://github.com/micropython/micropython/blob/b525f1c9ec8ffa9009754578932f3fad5f63026b/ports/rp2/modules/rp2.py#L52

The *actual* address of any given program is a function of the program offset + the instruction number. The `-1, -1` in the compiled programs above are their offsets for `PIO0` and `PIO1`. 

When you construct a `StateMachine`, Pico SDK's `_pio_add_program_at_offset` monkey-patches the instructions to add the program offset to each jump. That happens here: https://github.com/raspberrypi/pico-sdk/blob/f396d05f8252d4670d4ea05c8b7ac938ef0cd381/src/rp2_common/hardware_pio/pio.c#L104-L114

The `StateMachine` copies the program and uses that offset internally to set up the SM. Operative word being "internally", we can't get at it to use it within - for example - an `sm.exec("jmp(load)")` call.

But if we could modify `StateMachine` to accept an 9-element list including a labels dict, we could potentially support `jump` by having it pass that labels dict *back* into `asm_pio_encode` which it calls here:

https://github.com/micropython/micropython/blob/9e6885ad820a9f42bf93df56eb3f3be0c8639622/ports/rp2/rp2_pio.c#L685-L699

`asm_pio_encode` already jams a lot of state into the emitter, so the addition of labels is - I would guess? - not a huge leap:

https://github.com/micropython/micropython/blob/b525f1c9ec8ffa9009754578932f3fad5f63026b/ports/rp2/modules/rp2.py#L275-L283

We would have to copy and modify the dict to add the program offset to each instruction number, which constitutes some extra flash and memory usage. That makes me wonder if this edge case might have been deliberately avoided for sake of simplicity.

webdeb, 2023-04-24:
Yes, thats something what I had in mind, that it definitely must be possible, since the label is just an offset of the program, and the program offset (PC) is inside the statemachine. However, as the jmp instruction is a bit different from others (it requires an additional encoding) and since its almost impossible to predict the current state of the statemachine, a conditional jmp does not make sense from the outside of the statemachine anyway, I guess., I would request to add an **unconditional jmp** just as a separate method of the statemachine itself. 

The con is that it would be a separate method, some bytes of code. However, those bytes are very limited, and precise and would not overload the `rp2_state_machine_exec` method further, as it's kind of heavy already. Additionally it would not require to pass those metadata to the encoding functions, since we only need the pointers, which we would have in the python in `sm.labels` so the actual method would look something like (not a C code):

```
// pseudo code inspired by state_machine_restart :)
STATIC mp_obj_t rp2_state_machine_jm(mp_obj_t self_in, int label) {
    rp2_state_machine_obj_t *self = MP_OBJ_TO_PTR(self_in);
    pio_sm_exec(self->pio, self->sm, pio_encode_jmp(rp2_state_machine_initial_pc[self->id] + label));
    return mp_const_none;
}
```

And on the mp side:

```
sm.jmp(sm.labels["label"])
```

What do you think?

dpgeorge, 2023-05-18:
Commit f9958417d8adb21147de776361a4414073cf7c77 added the possibility to pass a pre-encoded instruction to `sm.exec(...)`.  Perhaps that can be used to solve your issue, by working out the exact jump instruction to pass in?

webdeb, 2023-07-08:
@dpgeorge good to know, thank you. I realized, that I am better with doing a conditional_jmp inside the program, as then I can control where the jumps happens (e.g. full loop). For me, it works by clearing the `isr` from outside and `jmp(not_y, "load")`

closed 2024-04-08 by webdeb