micropython/micropython #11284: f-string thousands separator doesn't work with floats

BrianPugh, 2023-04-17:
On v1.19.1, f-string thousands separator doesn't work with floats. Tested on both `unix` and `rp2040`:

Demo code:
```
foo_int = 123456
foo_float = 1234.56789
print(f"{foo_int:,}   {foo_float:,}")
```

CPython (correct):
```
123,456   1,234.56789
```

MicroPython (incorrect):
```
123,456   1234.57
```

dpgeorge, 2023-04-20:
Thanks for the report.  This is actually an issue with `str.format` rather than f-strings themselves.

Eg, on MicroPython:
```
'{:,}'.format(1234.)
'1234'
```
It should be `'1,234.0'`.

Gadgetoid, 2023-05-04:
Looks like `PF_FLAG_SHOW_COMMA` is correctly set but ignored in `mp_print_float` ( never passed to `mp_format_float` ).

I'd almost be tempted to special case floats to handle the integer and fractional portions separately, but there are sooooo many edge cases I can't even begin to have a go at it.