micropython/micropython #11313: docs: ports/unix/Readme.md: build instructions don't work / seem outdated

axel-kah, 2023-04-21:
**Environment:**
- git tag: `v1.19.1`
- `uname -a`
  ```bash
  Linux inspiron16p 5.15.90.1-microsoft-standard-WSL2 #1 SMP Fri Jan 27 02:56:13 UTC 2023 x86_64 x86_64 x86_64  GNU/Linux
  ```
- fully upgraded `Ubuntu 22.04.2 LTS`

**Problem:**

- List of build dependencies is missing `autoconf` (required by `libffi` when invoking `make deplibs`)
- Followed the build instructions but `./build-standard/micropython` does not exist. Instead `./build-standard` folder contains a lot of object files and the linked executable appears only in the unix ports root folder `./micropython` but seems to work just fine.
- `make deplibs` fails with a bunch of obsolete macro warnings and an undefined error macro:
  ```make
  Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
  cd ../../lib/libffi; ./autogen.sh
  autoreconf: export WARNINGS=
  autoreconf: Entering directory '.'
  autoreconf: configure.ac: not using Gettext
  autoreconf: running: aclocal --force -I m4
  autoreconf: configure.ac: tracing
  autoreconf: configure.ac: not using Libtool
  autoreconf: configure.ac: not using Intltool
  autoreconf: configure.ac: not using Gtkdoc
  autoreconf: running: /usr/bin/autoconf --force
  configure.ac:8: warning: The macro `AC_CANONICAL_SYSTEM' is obsolete.
  configure.ac:8: You should run autoupdate.
  ./lib/autoconf/general.m4:2081: AC_CANONICAL_SYSTEM is expanded from...
  configure.ac:8: the top level
  configure.ac:68: warning: The macro `AC_TRY_COMPILE' is obsolete.
  configure.ac:68: You should run autoupdate.
  ./lib/autoconf/general.m4:2847: AC_TRY_COMPILE is expanded from...
  lib/m4sugar/m4sh.m4:692: _AS_IF_ELSE is expanded from...
  lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
  ./lib/autoconf/general.m4:2249: AC_CACHE_VAL is expanded from...
  ./lib/autoconf/general.m4:2270: AC_CACHE_CHECK is expanded from...
  acinclude.m4:3: AC_FUNC_MMAP_BLACKLIST is expanded from...
  configure.ac:68: the top level
  configure.ac:85: warning: The macro `AC_HEADER_STDC' is obsolete.
  configure.ac:85: You should run autoupdate.
  ./lib/autoconf/headers.m4:704: AC_HEADER_STDC is expanded from...
  configure.ac:85: the top level
  configure.ac:112: warning: The macro `AC_TRY_COMPILE' is obsolete.
  configure.ac:112: You should run autoupdate.
  ./lib/autoconf/general.m4:2847: AC_TRY_COMPILE is expanded from...
  lib/m4sugar/m4sh.m4:692: _AS_IF_ELSE is expanded from...
  lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
  ./lib/autoconf/general.m4:2249: AC_CACHE_VAL is expanded from...
  ./lib/autoconf/general.m4:2270: AC_CACHE_CHECK is expanded from...
  m4/asmcfi.m4:1: GCC_AS_CFI_PSEUDO_OP is expanded from...
  configure.ac:112: the top level
  configure.ac:116: warning: The macro `AC_TRY_LINK' is obsolete.
  configure.ac:116: You should run autoupdate.
  ./lib/autoconf/general.m4:2920: AC_TRY_LINK is expanded from...
  lib/m4sugar/m4sh.m4:692: _AS_IF_ELSE is expanded from...
  lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
  ./lib/autoconf/general.m4:2249: AC_CACHE_VAL is expanded from...
  ./lib/autoconf/general.m4:2270: AC_CACHE_CHECK is expanded from...
  configure.ac:116: the top level
  configure.ac:132: warning: The macro `AC_TRY_COMPILE' is obsolete.
  configure.ac:132: You should run autoupdate.
  ./lib/autoconf/general.m4:2847: AC_TRY_COMPILE is expanded from...
  lib/m4sugar/m4sh.m4:692: _AS_IF_ELSE is expanded from...
  lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
  ./lib/autoconf/general.m4:2249: AC_CACHE_VAL is expanded from...
  ./lib/autoconf/general.m4:2270: AC_CACHE_CHECK is expanded from...
  configure.ac:132: the top level
  configure.ac:41: error: possibly undefined macro: AC_PROG_LIBTOOL
        If this token and others are legitimate, please use m4_pattern_allow.
        See the Autoconf documentation.
  autoreconf: error: /usr/bin/autoconf failed with exit status: 1
  make: *** [Makefile:322: ../../lib/libffi/configure] Error 1
  make: *** Deleting file '../../lib/libffi/configure'
  ```

I'd be happy to raise a PR fixing `README.md` but fixing the C-related build issue is not my area of expertise.

dlech, 2023-05-01:
On Ubuntu, I always use the system libffi instead of the submodule.

Happy to review a PR.

jonnor, 2024-07-13:
Is this still an issue with MicroPython 1.23 ?

closed 2024-09-06 by dpgeorge

agatti, 2024-09-06:
Seems like RV64 was left out from the supported platform lists :)

PR #15724 (2024-09-06): unix: Expand the build steps in the README.
### Summary

- Present the default build dependencies in one place at the top, and make a separate section about building standalone.

- Add steps for the "minimal" variant as well.

- Document that building standalone requires autoconf and libtool.

- Allow MICROPY_STANDALONE to be set as an environment variable (makes for easier documentation as well, as it has to be set for multiple make invocations).

Closes #11313

### Testing

I tested these changes by running various install and build steps in some throwaway Ubuntu 20.04 LTS containers (via `p [... cut]