micropython/micropython #11315: ESP32-S3 GPIO19/20 after disabling USB CDC

IcedRooibos, 2023-04-23:
I've built for `BOARD=GENERIC_S3_SPIRAM` with the following `sdkconfig.usb`

```
CONFIG_USB_ENABLED=n
CONFIG_USB_CDC_ENABLED=n
```

Firmware attached. This also has the fix from #11254 built-in.
[firmware-no-usb-rmt-loop-ok.zip](https://github.com/micropython/micropython/files/11303129/firmware-no-usb-rmt-loop-ok.zip)

However I can't actually set the values for pins 19 and 20 using micropython
```
>>> import machine
>>> a = machine.Pin(19, machine.Pin.OUT)
>>> a.value(1)
>>> a.value()
0
>>> b = machine.Pin(20, machine.Pin.OUT)
>>> b.value(0)
>>> b.value()
1
>>> a.on()
>>> a.value()
0
>>> b.off()
>>> b.value()
1
```

I've confirmed no hardware issues using an arduino sketch and testing with a voltmeter, the pins are high/low when they are supposed to be.

```
void setup() {
Serial.begin(115200);
pinMode(19, OUTPUT);
pinMode(20, OUTPUT);
pinMode(21, OUTPUT);

}

void loop() {
Serial.println("HIGH");
digitalWrite(19, HIGH);
digitalWrite(20, HIGH);
digitalWrite(21, HIGH);
delay(5000);
Serial.println("LOW");

digitalWrite(19, LOW);
digitalWrite(20, LOW);
digitalWrite(21, LOW);
delay(5000);

}
```

IcedRooibos, 2023-04-23:
My workaround for now is to use PWM and set the duty cycle to 100% or 0 depending on whether I need it on or off. A bit unfortunate to waste 3 PWM channels here but might be fine for now.

Curious what causes machine_pin_on to not work.

```
>>> import machine
>>> a = machine.PWM(machine.Pin(19, machine.Pin.OUT))
>>> a.duty_u16(0)
>>> a.duty_u16(65535)
```

robert-hh, 2023-04-23:
A duty cycle of  65535 return not exactly a continuous high level, at least at the last time I measured it. There is still a short low pulse at the output, duration 1/(f * 65535), minimum 12 ns, If just the average level is important, that's not a problem. But if the attached device responds to transitions, you may see unexpected behavior.

IcedRooibos, 2023-04-23:
Thanks Robert, that's great to know.

Do you have any guesses as to what might be causing this behavior with the pin.value() function not working as expected?

I added check_esp_err in the method that calls gpio_set_level [link](https://github.com/micropython/micropython/blob/master/ports/esp32/machine_pin.c#L388)
```
        check_esp_err(gpio_set_level(self->id, mp_obj_is_true(args[0])));
```

Firmware file:
[gpiodebugging.zip](https://github.com/micropython/micropython/files/11303748/gpiodebugging.zip)

But that didn't raise any errors either:
```
>>> import machine
>>> a = machine.Pin(19, machine.Pin.OUT)
>>> a.value(1)
>>> a.value()
0
>>> a.value(0)
>>> a.value()
0
```

robert-hh, 2023-04-23:
> Do you have any guesses as to what might be causing this behavior with the pin.value()

Actually not. Did you check:
- if in output mode, pin.value(x) does not set the output es expected using a multimeter?
- if in input mode and the pin manually set to low or high, pin.value() returns the proper value?

IcedRooibos, 2023-04-23:
Yes I checked with a multimeter to make sure the pins were indeed off. The voltage was ~0.3V for pin 19 and ~3.0V for pin 20.

In input mode it reads the value correctly. 0 first and then 1 after I manually connect it to 3.3V
```
>>> import time
>>> import machine
>>> a = machine.Pin(19, machine.Pin.IN)
>>> for i in range(0,10):
    a.value()
    time.sleep(1)
    
0
0
0
0
0
0
1
1
1
1
```

IcedRooibos, 2023-04-24:
Wow I've found an even weirder workaround.

If I declare the pin, initialize a PWM object using it, then reinitialize as a regular pin, it seems to work just fine.

```
>>> import machine
>>> a = machine.Pin(19, machine.Pin.OUT)
>>> a.value()
0
>>> a.value(1)
>>> a.value()
0
>>> a.value(1)
>>> a.value()
0
>>> a.value()
0
>>> a.value(1)
>>> a.value()
0
>>> a.value(1)
>>> a.value()
0
>>> a = machine.PWM(a)
>>> a = machine.Pin(19, machine.Pin.OUT)
>>> a.value()
1
>>> a.value(0)
>>> a.value()
0
>>> a.value(1)
>>> a.value()
1
>>> 
```

JoeSc, 2023-09-01:
FYI, I see similar issue where I cannot control GPIO19,GPIO20.  However your workarounds do not seem to work for me.  Did you ever find a solution?
I tried building micropython from source today and the issue persists

IcedRooibos, 2023-09-02:
Did you modify modify `sdkconfig.usb` to have the following?

```
CONFIG_USB_ENABLED=n
CONFIG_USB_CDC_ENABLED=n
```

What board are you using?

mortbauer, 2024-09-03:
@JoeSc did you manage to solve that issue, having exactly the same problem right now :(

projectgus, 2024-09-04:
Hi all, sorry I missed this issue until now. This is a limitation in the current firmware. There is a compile-time workaround described in #15769 that mostly works with the current MicroPython, but it should be easier to disable the USB pins both at runtime and at compile time.

Issue #14217 has the most detail about this, so I'm going to close this issue in favour of that one. Please subscribe there to get updates on this.

closed 2024-09-04 by projectgus

dev02prodeo, 2024-10-24:
> Wow I've found an even weirder workaround.
> 
> If I declare the pin, initialize a PWM object using it, then reinitialize as a regular pin, it seems to work just fine.
> 
> ```
> >>> import machine
> >>> a = machine.Pin(19, machine.Pin.OUT)
> >>> a.value()
> 0
> >>> a.value(1)
> >>> a.value()
> 0
> >>> a.value(1)
> >>> a.value()
> 0
> >>> a.value()
> 0
> >>> a.value(1)
> >>> a.value()
> 0
> >>> a.value(1)
> >>> a.value()
> 0
> >>> a = machine.PWM(a)
> >>> a = machine.Pin(19, machine.Pin.OUT)
> >>> a.value()
> 1
> >>> a.value(0)
> >>> a.value()
> 0
> >>> a.value(1)
> >>> a.value()
> 1
> >>> 
> ```
> 
> This also works if I deinit the PWM channel.
> 
> ```
> >>> a
> PWM(Pin(19))
> >>> a.deinit()
> >>> a
> PWM(Pin(19))
> >>> a.duty(100)
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
> RuntimeError: PWM inactive
> >>> a = machine.Pin(19, machine.Pin.OUT)
> >>> a.value()
> 1
> >>> a.value(0)
> >>> a.value()
> 0
> >>> a.value(1)
> >>> a.value()
> 1
> >>> 
> ```

This solved my problem too! I had a similar issue with ESP32-C3, having disabled USB_SERIAL_JTAG PORT and compiled micropython 1.24. It let me use the pins 18,19 (reserved for USB) without errors, but nothing really happened. After initialising them as PWM I can confirm that I can use them as GPIO or UART, as my intended use was. I cannot thank you enough!!